novatel/novatel_oem7_driver #1: ARM Builds are not supported

novatel-applications-engineering, 2020-04-29:
At this time, ARM64/ARMHF type builds of this driver are not supported.

Angusross111, 2020-07-09:
Is there any plans to support ARM systems in the future?

novatel-applications-engineering, 2020-07-09:
Hi @Angusross111, Thank you for asking. ARM support is possible in the future, but is not currently under active development. 

The original message for this Issue has been updated to now ask users to thumbs-up (+1) this Issue to help us get a sense of the level of interest or urgency. 

If you have a special business need for this, please email support@novatel.com describing your need.

venabled, 2020-08-20:
We deploy to ARM64 and would appreciate support for that platform.

SimkinTheWizard, 2020-09-15:
Hi, we too deploy on an ARM64 CPU, running an Ubuntu OS.

berkerlogoglu, 2020-09-30:
Many people use OEM7 on Nvidia Drive AGX, so it is really a big loss that you don't have ARM64 support.  We would love to see ARM64 support. 

Also if anyone has implemented the log decoder part, we would appreciate the help.

novatel-applications-engineering, 2020-10-01:
Thank you all for your comments, please continue to post your comments and especially provide notes on your specific ARM hardware and operating system. 

We are watching the feedback carefully and so far ARM support is by far the most requested next feature for this driver. 

We cannot commit to providing ARM support at this time, but it's possible that we may be able to assign the necessary development resources for this in the future. 

If you have an urgent request, please directly contact NovAtel Customer Service at support.novatel@hexagon.com

novatel-applications-engineering, 2021-01-29:
ARM support has been added by version 2.1.0.

closed 2021-01-29 by novatel-applications-engineering