novatel/novatel_oem7_driver #2: Are there any plans for releasing this in ROS2?

msmcconnell, 2020-05-28:
Are there any plans for releasing this driver in ROS2 and if so could you provide an approximate timeline or is external contribution necessary?

novatel-applications-engineering, 2020-05-29:
@msmcconnell Great question, thank you for raising it. 

The ROS2 implementation is currently under consideration but no specific timeline is available. 

Anyone who is also interested in this is invited to up-vote the original question above.

novatel-applications-engineering, 2021-04-30:
Update: a project to create the ROS2 implementation of novatel_oem7_driver is now underway. We a targeting a Q3 2021 release.

tgbosch, 2021-07-13:
Are there any news if you will be able to meet the Q3 2021 target?
Would also be interesting to know, if you will support a Windows-native ROS2 environment?

novatel-applications-engineering, 2021-07-19:
@tgbosch Yes, We are still on target for Q3 2021 release. 

A Windows-native ROS2 environment is on our radar, but there is no planned date for release.

Timple, 2021-08-13:
ROS2 support doesn't seem added yet, why was this closed?

novatel-applications-engineering, 2021-08-16:
@Timple Apologies! ROS2 is still on track for release. We will leave this open until the feature has been officially released.

novatel-applications-engineering, 2022-01-10:
[FYI we have officially released ROS2 support as of December 8, 2021](https://novatel.com/about-us/news-releases/hexagon-novatel-delivers-ros-2-compatible-drivers-to-the-oem7-receiver)

All necessary files are available here on Github.

closed 2022-01-10 by novatel-applications-engineering