novatel/novatel_oem7_driver #3: Feature Request: ROS Diagnostics

icolwell-as, 2020-06-25:
It would be handy if the driver published [diagnostic](http://wiki.ros.org/diagnostics) messages.

novatel-applications-engineering, 2020-07-06:
@icolwell-as, thank you for this request. Can you please describe what type of status info you would like to be published through the diagnostic package?

icolwell-as, 2020-07-08:
Here are a few ideas off the topic of my head:

- Publish topic status messages for each requested novatel message/ros topic ([diagnostic_updater](http://wiki.ros.org/diagnostic_updater) is useful for this). I believe something similar is already done by the driver and sent to rosout.
- A DiagnosticStatus message that represents the connection to the receiver (if it's connected or not)
- A DiagnosticStatus message for the current solution status, and other pertinent receiver status information like number of satellites tracked, etc.

Might also be a good idea to look at other opensource ROS drivers to get a feel for how they use diagnostics.
https://github.com/ros-drivers
I think a lot of the camera drivers use diagnostics.

novatel-applications-engineering, 2020-07-08:
Thank you @icolwell-as, we're looking in to this possibly being added to our next release of this driver.

Timple, 2020-09-22:
For us it would be useful to see IF rtcm data is flowing in. And IF this rtcm data is successfully applied.
This would be typical data for a diagnostics item.

MCFurry, 2021-04-22:
It would also be really helpful if the `status` field of the [NavSatFix](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html) message (/gps/fix) would be populated. This field is of type [NavSatStatus](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatStatus.html) and hence resembles the fix-status and gnss-services in use.
At this moment it is not and we have to get this information from other Novatel topics.