novatel/novatel_oem7_driver #4: Are there any plans for adding nmea_msgs/Sentence Message?

ctwu0314, 2020-06-29:
Because there are many self-driving software use nmea_msgs/Sentence Message.
(http://docs.ros.org/api/nmea_msgs/html/msg/Sentence.html)
Are there any plans for adding nmea_msgs/Sentence to publish.

novatel-applications-engineering, 2020-06-29:
Hello @ctwu0314, this is a great question. We're now discussing it internally and considering it for a future release of the novatel_oem7_driver.

The NMEA logs we support (such as GPGGA and GPRMC) are listed amongst many of our other logs on this page:
https://docs.novatel.com/OEM7/Content/Logs/Core_Logs.htm

Are there any specific OEM7-supported NMEA logs that you are interested in?

ctwu0314, 2020-07-02:
Hello
Thank you for your prompt reply.
I am interested in GPGGA , GPHDT , GPRMC.
You may be able to provide users to choose what NMEA information needs to be output.

novatel-applications-engineering, 2020-07-02:
@ctwu0314, okay, thank you for the input. It is being considered to be added to a future release of this driver. 

To anyone else interested in this - you are invited to thumbs-up the original ask to help us weigh the level of interest for this feature.

MinweiT, 2020-07-13:
Would love to see the NMEA sentences added to the msgs pool. In my experience, some devices and programs only accept NMEA formats. It would help a lot if the ROS driver can publish NMEA messages directly.

novatel-applications-engineering, 2020-07-13:
Thank you for your feedback @MinweiT. 

We are considering adapting the driver to publish NMEA formatted data under topics in the nmeg_msgs namespace as @ctwu0314 suggested. Would this satisfy your request too? If not, please describe what functionality would be most helpful to you in a little more detail please. 

Thank you again and have an excellent day.

MinweiT, 2020-07-14:
@novatel-applications-engineering 

Thank you for your notice. Besides @ctwu0314's recommendation, I would also like to have GPGST and GPVTG if possible.

novatel-applications-engineering, 2021-01-21:
As of ROS 2.0.0, All OEM7 NMEA sentences can be published as nmea_msgs/Sentence:
https://docs.novatel.com/OEM7/Content/Logs/NMEA_Standard_Logs_.htm?Highlight=nmea

Please note that only Odometry is supported.

closed 2021-01-21 by novatel-applications-engineering