novatel/novatel_oem7_driver #5: Possibility to add a range message or how to display it

Adabapt, 2020-07-22:
Hi!

Can it's possible or planned to add a range message to acces to this data.
At the start of the driver we can see that the RANGE log is activate but It can't be display.

Thanks for your driver and your job. I remain at your disposal.

novatel-applications-engineering, 2020-07-22:
Hi @Adabapt, thank you for your comment. You can configure the driver to log whatever additional logs you like. You can externally collect these logs or you can capture them through the /novatel/oem7/oem7raw topic. 

How and where would you prefer RANGE be displayed?

Adabapt, 2020-07-23:
I don't understand how work the oem7raw topic, It just displays a table of value with any explication.
Maybe this data can be available in an over the topic, with a new message type for example an oem7_range.msg.

Adabapt, 2020-07-24:
I have found a way to collect this data with the ros wiki.
For that, I created a bag and I filtered it to conserve only the oem7raw topic. After that, I converted it on NovatelConvert and I obtain a text file with the range data.

Thanks for your help, but I still think a topic for the range data and the others logs can be a good add.

novatel-applications-engineering, 2020-07-27:
Thank you @Adabapt for your input. We have a new feature request logged internally based on your feedback. I cannot comment if we'll be able to directly implement this support, but it will be discussed and considered. 

I'm glad you found the wiki and that it helped you. The wiki pages starting at http://wiki.ros.org/novatel_oem7_driver is intended to be the definitive guide for using novatel_oem7_driver.

closed 2022-01-10 by novatel-applications-engineering

carTloyal123, 2022-01-26:
Hey,

Was there any update on adding a range topic to the ROS implementation? I think this would be very handy for professional and research use to keep all data in rosbags instead of having to manage two sets of data.

Is there a way to add any additional topics besides the ones listed by default at the moment? I know you can remove them but is there a way to add any that are not already listed?

Thanks and look forward to hearing more.

novatel-applications-engineering, 2022-01-28:
Hi @carTloyal123, we currently have the feature in our backlog of items to add, we do not currently have a timetable on implementation of a RANGE topic. 

Our ROS Wiki has further information on how to modify the driver: https://wiki.ros.org/novatel_oem7_driver