novatel/novatel_oem7_driver #6: Attempt 0: timed out waiting for response

yucedagonurcan, 2020-07-28:
Hello,

We are experiencing timeout errors while using the oem7 ros driver.

Here is our oem7_net.launch file:
[image]

- You can see there is little change and it is for changing the IP. That is the same IP for connecting with the Novatel Web Interface.

Launching the launch file gives timeout errors:

`$ roslaunch launch/oem7_net.launch 
`
```
SUMMARY
========

PARAMETERS
 * /novatel/oem7/oem7_msgs/BDSEPHEMERIS: 1696
 * /novatel/oem7/oem7_msgs/BESTPOS: 42
 * /novatel/oem7/oem7_msgs/BESTUTM: 726
 * /novatel/oem7/oem7_msgs/BESTVEL: 99
 * /novatel/oem7/oem7_msgs/CORRIMUS: 2264
 * /novatel/oem7/oem7_msgs/GALFNAVEPHEMERIS: 1310
 * /novatel/oem7/oem7_msgs/GALINAVEPHEMERIS: 1309
 * /novatel/oem7/oem7_msgs/GLOEPHEMERIS: 723
 * /novatel/oem7/oem7_msgs/HEADING2: 1335
 * /novatel/oem7/oem7_msgs/IMURATECORRIMUS: 1362
 * /novatel/oem7/oem7_msgs/INSCONFIG: 1945
 * /novatel/oem7/oem7_msgs/INSPVAS: 508
 * /novatel/oem7/oem7_msgs/INSPVAX: 1465
 * /novatel/oem7/oem7_msgs/INSSTDEV: 2051
 * /novatel/oem7/oem7_msgs/INSUPDATESTATUS: 1825
 * /novatel/oem7/oem7_msgs/PSRDOP2: 1163
 * /novatel/oem7/oem7_msgs/RANGE: 43
 * /novatel/oem7/oem7_msgs/RAWEPHEM: 41
 * /novatel/oem7/oem7_msgs/RAWIMUSX: 1462
 * /novatel/oem7/oem7_msgs/RXSTATUS: 93
 * /novatel/oem7/oem7_msgs/TIME: 101
 * /novatel/oem7/receivers/main/HEADING2/frame_id: gps
 * /novatel/oem7/receivers/main/HEADING2/topic: /novatel/oem7/hea...
 * /novatel/oem7/receivers/main/NavSatFix/frame_id: gps
 * /novatel/oem7/receivers/main/NavSatFix/topic: /gps/fix
 * /novatel/oem7/receivers/main/oem7_if: Oem7ReceiverTcp
 * /novatel/oem7/receivers/main/oem7_ip_addr: 192.168.1.112
 * /novatel/oem7/receivers/main/oem7_msg_decoder: Oem7MessageDecoder
 * /novatel/oem7/receivers/main/oem7_msg_handlers: ['BESTPOSHandler'...
 * /novatel/oem7/receivers/main/oem7_port: 3001
 * /novatel/oem7/receivers/main/oem7_publish_unknown_oem7raw: True
 * /novatel/oem7/receivers/main/oem7_raw_msgs: ['BESTPOS', 'BEST...
 * /novatel/oem7/receivers/main/receiver_init_commands: ['UNLOGALL THISPO...
 * /novatel/oem7/supported_imus/0/name: Unknown
 * /novatel/oem7/supported_imus/0/rate: 0
 * /novatel/oem7/supported_imus/1/name: Honeywell HG1700 AG1
 * /novatel/oem7/supported_imus/1/rate: 100
 * /novatel/oem7/supported_imus/11/name: Honeywell HG1700 ...
 * /novatel/oem7/supported_imus/11/rate: 100
 * /novatel/oem7/supported_imus/12/name: Honeywell HG1700 ...
 * /novatel/oem7/supported_imus/12/rate: 100
 * /novatel/oem7/supported_imus/13/name: iMAR ilMU-FSAS
 * /novatel/oem7/supported_imus/13/rate: 200
 * /novatel/oem7/supported_imus/16/name: KVH CPT IMU
 * /novatel/oem7/supported_imus/16/rate: 200
 * /novatel/oem7/supported_imus/19/name: Northrop Grumman ...
 * /novatel/oem7/supported_imus/19/rate: 200
 * /novatel/oem7/supported_imus/20/name: Honeywell HG1930 ...
 * /novatel/oem7/supported_imus/20/rate: 100
 * /novatel/oem7/supported_imus/26/name: Northrop Grumman ...
 * /novatel/oem7/supported_imus/26/rate: 100
 * /novatel/oem7/supported_imus/27/name: Honeywell HG1900 ...
[... 263 lines cut ...]
[ERROR] [1595946404.489161253]: Attempt 0: timed out waiting for response.
[ERROR] [1595946407.489479761]: Attempt 1: timed out waiting for response.
[ERROR] [1595946410.489881098]: Attempt 2: timed out waiting for response.
[ERROR] [1595946413.490202973]: Attempt 3: timed out waiting for response.
[ERROR] [1595946416.490467308]: Attempt 4: timed out waiting for response.
[ERROR] [1595946419.490799234]: Attempt 5: timed out waiting for response.
[ERROR] [1595946422.491127299]: Attempt 6: timed out waiting for response.
[ERROR] [1595946425.491442513]: Attempt 7: timed out waiting for response.
[ERROR] [1595946428.491743454]: Attempt 8: timed out waiting for response.
[ERROR] [1595946431.492079993]: Attempt 9: timed out waiting for response.
[ERROR] [1595946431.492189096]: AACmd 'LOG RAWEPHEMB ONNEW' : ''
[ERROR] [1595946434.497087678]: Attempt 0: timed out waiting for response.
[ERROR] [1595946437.497389738]: Attempt 1: timed out waiting for response.
[ERROR] [1595946440.497696454]: Attempt 2: timed out waiting for response.
[ERROR] [1595946443.497973814]: Attempt 3: timed out waiting for response.
[ERROR] [1595946446.498318962]: Attempt 4: timed out waiting for response.
[ERROR] [1595946449.498666420]: Attempt 5: timed out waiting for response.
[ERROR] [1595946452.498965438]: Attempt 6: timed out waiting for response.
```

Here is our logging window:
[image]

Here is our position window:
[image]

novatel-applications-engineering, 2020-07-29:
@yucedagonurcan, thank you for reaching out. 

Based on what I see here, I think that the driver cannot connect to your receiver. It maybe that your port 3001 is already in use by something else. 

You could try updating your "oem7_port" in your launch file to be another value between 3001 and 3007. These are TCP ports (unless reconfigured on your receiver). These ports can only service one connection at a time. This approach could let whatever else is connected to your receiver continue, but presumably allow the ROS driver to use an unused port.

If that doesn't work, please next confirm that you can at least ping your receiver, for that you could use the command:
`ping 192.168.1.112`

yucedagonurcan, 2020-08-04:
Hello,

Changing port to `3002` from `3001` resolved the issue.

Thank you for your help.

closed 2020-08-04 by yucedagonurcan

jayjayhee, 2020-08-25:
Hi, 

I'm experiencing a similar issue using USB communication. The driver has been successfully configured, however there were several errors and there isn't any data in the /gps/imu and /corrimu topics. 
`jay@jay:~$ roslaunch novatel_oem7_driver oem7_tty.launch oem7_tty_name:=/dev/ttyUSB0
... logging to /home/jay/.ros/log/4e7d7940-e6ac-11ea-91c3-04d4c4070b48/roslaunch-jay-20192.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jay:44393/

SUMMARY
========

PARAMETERS
 * /novatel/oem7/oem7_msgs/BDSEPHEMERIS: 1696
 * /novatel/oem7/oem7_msgs/BESTPOS: 42
 * /novatel/oem7/oem7_msgs/BESTUTM: 726
 * /novatel/oem7/oem7_msgs/BESTVEL: 99
 * /novatel/oem7/oem7_msgs/CORRIMUS: 2264
 * /novatel/oem7/oem7_msgs/GALFNAVEPHEMERIS: 1310
 * /novatel/oem7/oem7_msgs/GALINAVEPHEMERIS: 1309
 * /novatel/oem7/oem7_msgs/GLOEPHEMERIS: 723
 * /novatel/oem7/oem7_msgs/HEADING2: 1335
 * /novatel/oem7/oem7_msgs/IMURATECORRIMUS: 1362
 * /novatel/oem7/oem7_msgs/INSCONFIG: 1945
 * /novatel/oem7/oem7_msgs/INSPVAS: 508
 * /novatel/oem7/oem7_msgs/INSPVAX: 1465
 * /novatel/oem7/oem7_msgs/INSSTDEV: 2051
 * /novatel/oem7/oem7_msgs/INSUPDATESTATUS: 1825
 * /novatel/oem7/oem7_msgs/PSRDOP2: 1163
 * /novatel/oem7/oem7_msgs/RANGE: 43
 * /novatel/oem7/oem7_msgs/RAWEPHEM: 41
 * /novatel/oem7/oem7_msgs/RAWIMUSX: 1462
 * /novatel/oem7/oem7_msgs/RXSTATUS: 93
 * /novatel/oem7/oem7_msgs/TIME: 101
 * /novatel/oem7/receivers/main/BESTPOS/frame_id: gps
 * /novatel/oem7/receivers/main/BESTPOS/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/BESTUTM/frame_id: gps
 * /novatel/oem7/receivers/main/BESTUTM/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/BESTVEL/frame_id: gps
 * /novatel/oem7/receivers/main/BESTVEL/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/CORRIMU/frame_id: gps
 * /novatel/oem7/receivers/main/CORRIMU/topic: /novatel/oem7/cor...
 * /novatel/oem7/receivers/main/GPSFix/frame_id: gps
 * /novatel/oem7/receivers/main/GPSFix/topic: /gps/gps
 * /novatel/oem7/receivers/main/HEADING2/frame_id: gps
 * /novatel/oem7/receivers/main/HEADING2/topic: /novatel/oem7/hea...
 * /novatel/oem7/receivers/main/IMU/frame_id: gps
 * /novatel/oem7/receivers/main/IMU/topic: /gps/imu
 * /novatel/oem7/receivers/main/INSCONFIG/frame_id: gps
 * /novatel/oem7/receivers/main/INSCONFIG/queue_size: 10
 * /novatel/oem7/receivers/main/INSCONFIG/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/INSPVA/frame_id: gps
 * /novatel/oem7/receivers/main/INSPVA/topic: /novatel/oem7/inspva
 * /novatel/oem7/receivers/main/INSPVAX/frame_id: gps
 * /novatel/oem7/receivers/main/INSPVAX/queue_size: 10
 * /novatel/oem7/receivers/main/INSPVAX/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/INSSTDEV/frame_id: gps
 * /novatel/oem7/receivers/main/INSSTDEV/queue_size: 10
 * /novatel/oem7/receivers/main/INSSTDEV/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/NavSatFix/frame_id: gps
 * /novatel/oem7/receivers/main/NavSatFix/topic: /gps/fix
 * /novatel/oem7/receivers/main/Oem7RawMsg/frame_id: gps
 * /novatel/oem7/receivers/main/Oem7RawMsg/queue_size: 200
 * /novatel/oem7/receivers/main/Oem7RawMsg/topic: /novatel/oem7/oem...
 * /novatel/oem7/receivers/main/RXSTATUS/frame_id: gps
 * /novatel/oem7/receivers/main/RXSTATUS/queue_size: 10
 * /novatel/oem7/receivers/main/RXSTATUS/topic: /novatel/oem7/rxs...
 * /novatel/oem7/receivers/main/TIME/frame_id: gps
[... 136 lines cut ...]
[ INFO] [1598343835.275313073]: AACmd 'LOG TIMEB ONTIME 1' : 'OK'
[ INFO] [1598343835.283167604]: AACmd 'LOG RAWIMUSXB ONNEW' : 'OK'
[ INFO] [1598343835.288881066]: AACmd 'LOG INSUPDATESTATUSB ONNEW' : 'OK'
[ INFO] [1598343835.295749052]: AACmd 'LOG RAWEPHEMB ONNEW' : 'OK'
[ INFO] [1598343835.301847576]: AACmd 'LOG GLOEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598343835.309146055]: AACmd 'LOG BDSEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598343835.320089624]: AACmd 'LOG GALFNAVEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598343835.326103844]: AACmd 'LOG GALINAVEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598343835.334126694]: AACmd 'LOG RANGEB ONTIME 1' : 'OK'
[ INFO] [1598343835.334782293]: Oem7 configuration completed.
^C[novatel/oem7/receivers/main/config-4] killing on exit
[novatel/oem7/receivers/main-3] killing on exit
[ INFO] [1598343846.146462017]: Unloading nodelet /novatel/oem7/receivers/main/config from manager /novatel/oem7/driver
[novatel/oem7/driver-2] killing on exit
[ INFO] [1598343846.149899546]: Unloading nodelet /novatel/oem7/receivers/main from manager /novatel/oem7/driver
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
It can be seen that there are 3 errors reported and they might be the reason why I can't get any imu data.
`[ERROR] [1598343835.180147191]: Attempt 0: timed out waiting for response.`
`[ERROR] [1598343835.196250946]: AACmd 'LOG INSCONFIGB ONTIME 300' : 'ERROR:Trigger ONTIME not valid for this log'`
`[ERROR] [1598343835.260403646]: AACmd 'LOG CORRIMUSB ONTIME 0.01' : 'ERROR:Invalid Message. Field = 2'`

jayjayhee, 2020-08-25:
Hi

I'm experiencing a similar issue when using usb operation method. There were 3 error messages reported as follows:

`[ERROR] [1598343835.180147191]: Attempt 0: timed out waiting for response.`
`[ERROR] [1598343835.196250946]: AACmd 'LOG INSCONFIGB ONTIME 300' : 'ERROR:Trigger ONTIME not valid for this log'`
`[ERROR] [1598343835.260403646]: AACmd 'LOG CORRIMUSB ONTIME 0.01' : 'ERROR:Invalid Message. Field = 2'`

Additionally, although the /gps/imu and /corrimu/ topics are published, there isn't any imu data under these topics.