novatel/novatel_oem7_driver #7: Support for other OEM devices

aishanou-farmwise, 2020-08-05:
Can this be used for other OEM devices such as OEMStar? I specifically want to use this for AG-STAR-10HZ-D-G received.

novatel-applications-engineering, 2020-08-05:
@aishanou-farmwise, the OEMStar has not been tested with the novatel_oem7_driver for ROS. However, by inspecting the _std_init_commands.yml_ file, we can see that the _GPSFix_ and _NavSatFix_ topics require _BESTPOSB_ and _BESTVELB_, which are both provided by the OEMStar, as indicated in the OEMStar Firmware Reference Manual, [available here](https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Manuals/om-20000127/om-20000127.pdf).

Therefore the take-away is that it's possible that it can work, but we haven't tested this configuration yet. 

We can look in to requisitioning the necessary hardware and trying this, then responding here. Does that work for you?

aishanou-farmwise, 2020-08-06:
Yes that would be great. We will also test on our end then and see if it works out of the box.

novatel-applications-engineering, 2020-08-06:
@aishanou-farmwise I've now run up our ROS driver, briefly against an OEMStar receiver and for topics like /gps/fix which only depend on older style logs like BESTPOS and BESTVEL, it appeared to function. I observed positions and time measurements available through the ROS driver from an OEMStar. 

Note that if you run it up with the default logging configuration, there will be various error messages due to the driver requesting unsupported logs.

So, the driver can work with an OEMStar. I want to state that this is not a tested configuration and it may not be stable. Future versions of this driver might not work with OEMStar type receivers. However, given how this works, I think basic topics are likely to work longer term since the required logs and their formats are not likely to change.

Do you have any further questions @aishanou-farmwise?

aishanou-farmwise, 2020-08-07:
Thank you for trying it out and I am glad we can use it. One last question `oem7_tty.launch` is the launch file to be used with the RS232 connected receiver right? This is the one we use for AG-STAR-10HZ-D-G

aishanou-farmwise, 2020-08-08:
I tried launching with `oem7_tty.launch` and as you mentioned I am seeing a bunch of errors for unsupported logs. However I am not seeing any messages on the `/gps/fix` , `/gps/gps` , `/novatel/oem7/bestpos` and `/novatel/oem7/bestvel` topics. The nodelet eventually stops due to the errors. I have verified that the port is opened based on these logs. 

`[ INFO] [/novatel/oem7/driver] [52] [1596849751.655285651]: Oem7SerialPort['/dev/NOVATELGPS' : 9600]
[ INFO] [/novatel/oem7/driver] [62] [1596849751.655589097]: Oem7SerialPort open: '1; err: system:0
[ INFO] [/novatel/oem7/driver] [68] [1596849751.655654664]: Oem7SerialPort set_option baud_rate: '9600 : system:0
[ERROR] [/novatel/oem7/driver] [144] [1596849751.655705741]: Oem7Receiver: read error: 2; endpoint open: 1 errors/max: 4/7
[ERROR] [/novatel/oem7/driver] [95] [1596849751.655853482]: Oem7Receiver: close error: 0
[ INFO] [/novatel/oem7/driver] [52] [1596849752.659801131]: Oem7SerialPort['/dev/NOVATELGPS' : 9600]
[ INFO] [/novatel/oem7/driver] [62] [1596849752.660114964]: Oem7SerialPort open: '1; err: system:0
[ INFO] [/novatel/oem7/driver] [68] [1596849752.660192547]: Oem7SerialPort set_option baud_rate: '9600 : system:0
[ERROR] [/novatel/oem7/driver] [144] [1596849752.660257779]: Oem7Receiver: read error: 2; endpoint open: 1 errors/max: 5/7
[ERROR] [/novatel/oem7/driver] [95] [1596849752.660429626]: Oem7Receiver: close error: 0
[ INFO] [/novatel/oem7/driver] [52] [1596849753.664239974]: Oem7SerialPort['/dev/NOVATELGPS' : 9600]
[ INFO] [/novatel/oem7/driver] [62] [1596849753.664581446]: Oem7SerialPort open: '1; err: system:0
[ INFO] [/novatel/oem7/driver] [68] [1596849753.664657315]: Oem7SerialPort set_option baud_rate: '9600 : system:0
[ERROR] [/novatel/oem7/driver] [144] [1596849753.664721091]: Oem7Receiver: read error: 2; endpoint open: 1 errors/max: 6/7
[ERROR] [/novatel/oem7/driver] [95] [1596849753.664868623]: Oem7Receiver: close error: 0
[ INFO] [/novatel/oem7/driver] [52] [1596849754.668708899]: Oem7SerialPort['/dev/NOVATELGPS' : 9600]
[ INFO] [/novatel/oem7/driver] [62] [1596849754.669018706]: Oem7SerialPort open: '1; err: system:0
[ INFO] [/novatel/oem7/driver] [68] [1596849754.669094156]: Oem7SerialPort set_option baud_rate: '9600 : system:0
[ERROR] [/novatel/oem7/driver] [144] [1596849754.669154948]: Oem7Receiver: read error: 2; endpoint open: 1 errors/max: 7/7
[ERROR] [/novatel/oem7/driver] [95] [1596849754.669301068]: Oem7Receiver: close error: 0
[ERROR] [/novatel/oem7/driver] [62] [1596849755.669538907]: Oem7Receiver: Max Num IO errors exceeded: 7
[ WARN] [/novatel/oem7/driver] [108] [1596849755.669688047]: Decoder: no more messages available.
[ INFO] [/novatel/oem7/driver] [248] [1596849755.669782589]: Log Statistics:
[ INFO] [/novatel/oem7/driver] [250] [1596849755.669863998]: Logs: 27; unknown: 11; discarded: 0
[ INFO] [/novatel/oem7/driver] [259] [1596849755.669943591]: Log[](0):2
[ INFO] [/novatel/oem7/driver] [259] [1596849755.670006600]: Log[BESTPOS](42):16
[ INFO] [/novatel/oem7/driver] [259] [1596849755.670068633]: Log[](93):1
[ INFO] [/novatel/oem7/driver] [259] [1596849755.670123981]: Log[BESTVEL](99):8
[ WARN] [/novatel/oem7/driver] [381] [1596849755.670181277]: No more input from Decoder; Oem7MessageNodelet finished.
`
What do I need to do to obtain messages on the above listed topics?

novatel-applications-engineering, 2020-08-10:
@aishanou-farmwise - When I ran my quick test, I used the [rosbag](http://wiki.ros.org/rosbag) command to record the data from novatel_oem7_driver. The command I used looked like:
`rosbag record -a`

There are use examples of this for the novatel_oem7_driver wiki documentation, [here](https://wiki.ros.org/novatel_oem7_driver/runtime_operation#Recording_data_to_a_ROS_bag).

Since your issue appears to be getting specific to your particular case, please create a support case with NovAtel Applications Engineering, we'll continue support from there rather than keeping it public in GitHub. To create a support case with NovAtel Applications Engineering, please email: support.novatel@hexagon.com

closed 2020-08-10 by novatel-applications-engineering

aishanou-farmwise, 2020-08-10:
Thank you for you response! I have followed the instructions listed on the wiki documentation. I will email for more details.