novatel/novatel_oem7_driver #10: No data in /gps/imu topic

jayjayhee, 2020-08-25:
Hi

I'm experiencing an issue when using usb operation method. There were 3 error messages reported as follows:

[ERROR] [1598343835.180147191]: Attempt 0: timed out waiting for response.
[ERROR] [1598343835.196250946]: AACmd 'LOG INSCONFIGB ONTIME 300' : 'ERROR:Trigger ONTIME not valid for this log'
[ERROR] [1598343835.260403646]: AACmd 'LOG CORRIMUSB ONTIME 0.01' : 'ERROR:Invalid Message. Field = 2'

the complete command window is shown below:

`roslaunch novatel_oem7_driver oem7_tty.launch oem7_tty_name:=/dev/ttyUSB0
... logging to /home/jay/.ros/log/6e23324e-e6b6-11ea-91c3-04d4c4070b48/roslaunch-jay-29462.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jay:35349/

SUMMARY
========

PARAMETERS
 * /novatel/oem7/oem7_msgs/BDSEPHEMERIS: 1696
 * /novatel/oem7/oem7_msgs/BESTPOS: 42
 * /novatel/oem7/oem7_msgs/BESTUTM: 726
 * /novatel/oem7/oem7_msgs/BESTVEL: 99
 * /novatel/oem7/oem7_msgs/CORRIMUS: 2264
 * /novatel/oem7/oem7_msgs/GALFNAVEPHEMERIS: 1310
 * /novatel/oem7/oem7_msgs/GALINAVEPHEMERIS: 1309
 * /novatel/oem7/oem7_msgs/GLOEPHEMERIS: 723
 * /novatel/oem7/oem7_msgs/HEADING2: 1335
 * /novatel/oem7/oem7_msgs/IMURATECORRIMUS: 1362
 * /novatel/oem7/oem7_msgs/INSCONFIG: 1945
 * /novatel/oem7/oem7_msgs/INSPVAS: 508
 * /novatel/oem7/oem7_msgs/INSPVAX: 1465
 * /novatel/oem7/oem7_msgs/INSSTDEV: 2051
 * /novatel/oem7/oem7_msgs/INSUPDATESTATUS: 1825
 * /novatel/oem7/oem7_msgs/PSRDOP2: 1163
 * /novatel/oem7/oem7_msgs/RANGE: 43
 * /novatel/oem7/oem7_msgs/RAWEPHEM: 41
 * /novatel/oem7/oem7_msgs/RAWIMUSX: 1462
 * /novatel/oem7/oem7_msgs/RXSTATUS: 93
 * /novatel/oem7/oem7_msgs/TIME: 101
 * /novatel/oem7/receivers/main/BESTPOS/frame_id: gps
 * /novatel/oem7/receivers/main/BESTPOS/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/BESTUTM/frame_id: gps
 * /novatel/oem7/receivers/main/BESTUTM/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/BESTVEL/frame_id: gps
 * /novatel/oem7/receivers/main/BESTVEL/topic: /novatel/oem7/bes...
 * /novatel/oem7/receivers/main/CORRIMU/frame_id: gps
 * /novatel/oem7/receivers/main/CORRIMU/topic: /novatel/oem7/cor...
 * /novatel/oem7/receivers/main/GPSFix/frame_id: gps
 * /novatel/oem7/receivers/main/GPSFix/topic: /gps/gps
 * /novatel/oem7/receivers/main/HEADING2/frame_id: gps
 * /novatel/oem7/receivers/main/HEADING2/topic: /novatel/oem7/hea...
 * /novatel/oem7/receivers/main/IMU/frame_id: gps
 * /novatel/oem7/receivers/main/IMU/topic: /gps/imu
 * /novatel/oem7/receivers/main/INSCONFIG/frame_id: gps
 * /novatel/oem7/receivers/main/INSCONFIG/queue_size: 10
 * /novatel/oem7/receivers/main/INSCONFIG/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/INSPVA/frame_id: gps
 * /novatel/oem7/receivers/main/INSPVA/topic: /novatel/oem7/inspva
 * /novatel/oem7/receivers/main/INSPVAX/frame_id: gps
 * /novatel/oem7/receivers/main/INSPVAX/queue_size: 10
 * /novatel/oem7/receivers/main/INSPVAX/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/INSSTDEV/frame_id: gps
 * /novatel/oem7/receivers/main/INSSTDEV/queue_size: 10
 * /novatel/oem7/receivers/main/INSSTDEV/topic: /novatel/oem7/ins...
 * /novatel/oem7/receivers/main/NavSatFix/frame_id: gps
 * /novatel/oem7/receivers/main/NavSatFix/topic: /gps/fix
[... 130 lines cut ...]
[ INFO] [1598348183.296117783]: AACmd 'LOG INSCONFIGB ONCE' : 'OK'
[ INFO] [1598348183.300674495]: IMU: 'Epson G320N', rate= 125
[ERROR] [1598348183.302643357]: AACmd 'LOG INSCONFIGB ONTIME 300' : 'ERROR:Trigger ONTIME not valid for this log'
[ INFO] [1598348183.309520697]: AACmd 'LOG RXSTATUSB ONCHANGED' : 'OK'
[ INFO] [1598348183.315281163]: AACmd 'LOG BESTPOSB ONTIME 0.1' : 'OK'
[ INFO] [1598348183.323326704]: AACmd 'LOG BESTVELB ONTIME 0.1' : 'OK'
[ INFO] [1598348183.331940016]: AACmd 'LOG BESTUTMB ONTIME 1' : 'OK'
[ INFO] [1598348183.338498924]: AACmd 'LOG HEADING2B ONNEW' : 'OK'
[ INFO] [1598348183.345930291]: AACmd 'LOG PSRDOP2B ONCHANGED' : 'OK'
[ INFO] [1598348183.352300335]: AACmd 'LOG INSPVASB ONTIME 0.02' : 'OK'
[ INFO] [1598348183.358375687]: AACmd 'LOG INSPVAXB ONTIME 1' : 'OK'
[ERROR] [1598348183.365567507]: AACmd 'LOG CORRIMUSB ONTIME 0.01' : 'ERROR:Invalid Message. Field = 2'
[ INFO] [1598348183.371581626]: AACmd 'LOG INSSTDEVB ONTIME 1' : 'OK'
[ INFO] [1598348183.378204374]: AACmd 'LOG TIMEB ONTIME 1' : 'OK'
[ INFO] [1598348183.387402979]: AACmd 'LOG RAWIMUSXB ONNEW' : 'OK'
[ INFO] [1598348183.395132850]: AACmd 'LOG INSUPDATESTATUSB ONNEW' : 'OK'
[ INFO] [1598348183.402163923]: AACmd 'LOG RAWEPHEMB ONNEW' : 'OK'
[ INFO] [1598348183.410159415]: AACmd 'LOG GLOEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598348183.417501333]: AACmd 'LOG BDSEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598348183.441648973]: AACmd 'LOG GALFNAVEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598348183.450066874]: AACmd 'LOG GALINAVEPHEMERISB ONNEW' : 'OK'
[ INFO] [1598348183.460385374]: AACmd 'LOG RANGEB ONTIME 1' : 'OK'
[ INFO] [1598348183.460781212]: Oem7 configuration completed.`

Additionally, although configuration is completed despite errors and the /gps/imu and /corrimu/ topics are published, there isn't any imu data under these topics.

jayjayhee, 2020-08-25:
I actually don't consider this as a bug but somehow it shows a label of bug. I guess I misclicked something...

novatel-applications-engineering, 2020-08-25:
Hi @jayjayhee,
The settings of novatel_oem7_driver must match the abilities of the receiver being used with the driver. I suspect that your current novatel_oem7_driver settings are not compatible with your receiver.

This should be handled as a support case with NovAtel Customer Support. You may start a case by reaching out to us at support.novatel@hexagon.com. I'll watch for the case and take it when it comes in. Please include any customized .yaml files. Once the support case is open I'll send you a link to a location where you can also upload sample data such as bag files.

Thank you, talk to you soon,
NovAtel Customer Support

closed 2020-08-25 by novatel-applications-engineering

jayjayhee, 2020-08-26:
Thanks for your response. I really appreciate it.

Daniel-git-Hub-2021, 2021-03-27:
Hy @jayjayhee, 

I am having the same problem as you, would it be possible to say what you did to get around this situation? 

I’m stuck with a few days to solve this, since the data published in gps / imu or novatel / oem7 / corrimu are zeroed.

Tank You!

juhe9842, 2021-03-29:
Hi @Daniel-git-Hub-2021 

It's been quite a while since the time I solved this problem so I can barely remember the details I've been through. But I will try to make it clear to you. Those errors are mainly because the firmware of the imu doesn't match this driver. This driver has been updated for the new firmware and I was usin the old one. The best solution I would suggest is to update the firmware of the imu. If it's not available, you can also modify this driver based on the user manual to get the data you want. According to my memory, the topic "CORRIMUDATA" has been modified so you need to change it back to the old version (Compare the two versions of manual to see what has been changed).

Daniel-git-Hub-2021, 2021-03-29:
@juhe9842 

Thank you for the information! I will calmly analyze the imu driver that is currently active! Your response was of great help. 

have a good week.

novatel-applications-engineering, 2021-03-30:
Hi @Daniel-git-Hub-2021,

As mentioned in the response above, the settings of novatel_oem7_driver must match the OEM7 receiver and IMU (or combined OEM7 receiver + IMU enclosure product) being used with the driver. I suspect that your current novatel_oem7_driver settings are not compatible with your receiver. For example, you are using an OEM6 generation product or a product with an unsupported IMU. 

If you are unable to resolve the issue on your own, we'd be happy to assist through a support case with NovAtel Customer Support. You may start a case by reaching out to us at support.novatel@hexagon.com. We'll watch for the case and take it when it comes in. Please include any customized .yaml files. 

Thank you,
NovAtel Customer Support

Daniel-git-Hub-2021, 2021-03-31:
Hi, @novatel-applications-engineering,

My Novatel receiver is an OEM7 (PwrPak7D-E1) with integrated IMON EPSON G320N. I believe it is something related to SPAN initialization. Every time I turn the receiver off, it stays OFF. 

In addition, the INS light is only flashing yellow, it does not stabilize for anything. Would you have any tips on how I can proceed?

Grateful for your help.

Daniel-git-Hub-2021, 2021-03-31:
Hi, @novatel-applications-engineering,

My Novatel receiver is an OEM7 (PwrPak7D-E1) with integrated IMU EPSON G320N. I believe it is something related to SPAN initialization. Every time I turn the receiver off, it stays OFF.

In addition, the INS light is only flashing yellow, it does not stabilize for anything. Would you have any tips on how I can proceed?

Grateful for your help.

Daniel-git-Hub-2021, 2021-04-01:
@novatel-applications-engineering, 

I believe I found out where the problem is.

I need to modify the sdt_init_commands.yaml file where I will inform the imu data correctly as: 

"CONNECTIMU SPI EPSON_G320"

"SETINSTRANSLATION ANT1"

"SETINSTRANSLATION ANT2"

In addition to the rotations.

However, I don't know exactly how to make these changes to this file.

When I did this for "NovAtelApplicationSuite" it worked and the INS light stabilized in green and I got information from the IMU.

However, when I turn off the Receiver these settings are cleared and then I would have to enter NovAtelApplicationSuite again and set everything up again.

I believe that if configured correctly, when I initialize the file through ROS this initialization / configuration will occur correctly.

Could you help me edit this file?

Grateful for the attention, and I look forward to your return.

novatel-applications-engineering, 2021-04-05:
Hi @Daniel-git-Hub-2021,
We have responded to you through the NovAtel support queue. Full details and examples of modifying the receiver configuration via the std_init_commands.yaml file can be found here: [http://wiki.ros.org/novatel_oem7_driver/configuration](http://wiki.ros.org/novatel_oem7_driver/configuration)

MCFurry, 2021-04-19:
Just for reference to anyone who is reading this topic.

I recently connected a SMART7-S receiver to this driver and also did not have any `/gps/imu` messages published.

I tried manually executing the `LOG CORRIMUSB ONTIME 0.01` command which resulted in an error message. 
Then I found:
[image]
ref: https://docs.novatel.com/OEM7/Content/SPAN_Logs/CORRIMUDATA.htm
And enabling CORRIMUDATA log did work!

Hence I concluded that the firmware on my receiver was too old to contain the CORRIMUS log that this ROS driver is requesting by default. Upgrading the receiver firmware indeed fixed that!

ajay1606, 2021-09-17:
@novatel-applications-engineering , 
Hello, thanks for sharing information here, Indeed i was also stuck with same issue. After going through all your comments i realized root cause of issue. It was very much helpful in understanding actual cause.

But unfortunately, i couldn't able to fix the issue. We have two units of PwrPak7D-E1 receivers.  

**Receiver 1:** 
[image]
Using Receiver 1: am able to run novatel_oem7_driver without any issues.

**Receiver 2:**
[image]
With receiver 2 am getting following error message continously.

**Error Message:** 
[image]

We greatly appreciate if you could mention specific setting that needs to configure. 

Regards,
Ajay

novatel-applications-engineering, 2021-09-23:
Thank you for reaching out @ajay1606! It looks like a case was opened up in the NovAtel support queue on your issue and our team will help you out there.

programath, 2021-10-16:
> @novatel-applications-engineering , Hello, thanks for sharing information here, Indeed i was also stuck with same issue. After going through all your comments i realized root cause of issue. It was very much helpful in understanding actual cause.
> 
> But unfortunately, i couldn't able to fix the issue. We have two units of PwrPak7D-E1 receivers.
> 
> **Receiver 1:** [image] Using Receiver 1: am able to run novatel_oem7_driver without any issues.
> 
> **Receiver 2:** [image] With receiver 2 am getting following error message continously.
> 
> **Error Message:** [image]
> 
> We greatly appreciate if you could mention specific setting that needs to configure to fix the this error message.
> 
> Regards, Ajay

Hi Ajay,

Seems I encountered the same issue with yours.  Have you resolved it?