novatel/novatel_oem7_driver #15: Namespaces not supported

Timple, 2021-03-23:
**Describe the bug**
Using the launchfiles provided, it is not possible to use namespaces.

**To Reproduce**
```
roslaunch novatel_oem7_driver oem7_net.launch oem7_ip_addr:=1.1.1.1 __ns:=myrobot
```

**Current behavior**
```

SUMMARY
========

PARAMETERS
 * /novatel/oem7/oem7_msgs/BDSEPHEMERIS: 1696
 * /novatel/oem7/oem7_msgs/BESTPOS: 42
 * /novatel/oem7/oem7_msgs/BESTUTM: 726
 * /novatel/oem7/oem7_msgs/BESTVEL: 99
...
process[novatel/oem7/driver-1]: started with pid [333625]
...
```
**Expected behavior**
```

SUMMARY
========

PARAMETERS
 * /myrobot/novatel/oem7/oem7_msgs/BDSEPHEMERIS: 1696
 * /myrobot/novatel/oem7/oem7_msgs/BESTPOS: 42
 * /myrobot/novatel/oem7/oem7_msgs/BESTUTM: 726
 * /myrobot/novatel/oem7/oem7_msgs/BESTVEL: 99
...
process[/myrobot/novatel/oem7/driver-1]: started with pid [333625]
...
```
(Note the extra `/myrobot` at the beginning of the parameters and node names.

**Environment (please complete the following information):**
 - OS: ubuntu:20.04
 - Browser: How is this relevant?
 - Version: 2.2.0

**Additional context**
Solution should be easy by removing all prepending slashes.
And maybe `novatel_oem7` is less namespace-depth than `novatel/oem7`?

novatel-applications-engineering, 2021-03-25:
Hi @Timple. Please create a support case with NovAtel Customer Support, we'll continue support from there rather than keeping it public in GitHub. To create a support case with NovAtel Customer Support, please email: support.novatel@hexagon.com with 'https://github.com/novatel/novatel_oem7_driver/issues/15' mentioned in the body of the email, and we will merge the existing information you already provided. Thank you!

novatel-applications-engineering, 2021-04-20:
We confirmed the issue and have added this to our backlog. 

As a quick-fix solution:
As users have all the sources, if this becomes an urgent issue, you can make this namespace change yourself as quick solution.

Timple, 2021-11-18:
So this issue cost us (and four Novatel engineers) half a day yesterday.

Would you mind keeping it open until it is actually solved?

(We didn't solve it yet because it isn't a daily issue for us, but this is probably why it's good to solve it anyway :slightly_smiling_face: )

novatel-applications-engineering, 2022-01-12:
Apologies for the opening and closing of this issue. We have added it to our backlog of issues to fix and will implement in future releases.

closed 2022-01-12 by novatel-applications-engineering

Timple, 2022-01-13:
For transparency reasons, wouldn't it be easier to keep issues open until resolved?