novatel/novatel_oem7_driver #17: Status field of NavSatFix message is not populated

MCFurry, 2021-04-22:
It would be really helpful if the `status` field of the [NavSatFix](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html) message (/gps/fix) would actually be populated. 
This field is of type [NavSatStatus](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatStatus.html) and hence resembles the fix-status and gnss-services in use.
At this moment it is not and we have to get this information from other Novatel topics.

novatel-applications-engineering, 2021-05-03:
Hi @MCFurry thank you for your comment. To clarify, right now there is no value of any type populating the [NavSatFix](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatStatus.html) Status field?

MCFurry, 2021-05-03:
Exactly, it seems that the other fields [are filled](https://github.com/novatel/novatel_oem7_driver/blob/5ae6c1671b465668aba11c27a4bd7b32968f0fc3/src/novatel_oem7_driver/src/bestpos_handler.cpp#L260), except for the Status struct.
ROS will therefore create the message with default field-values of zero. 
So in this case, the driver is actually claiming to always have an 'unaugmented fix' without any gps services used (gps/glonass/glaileo) as per definition of the [NavSatStatus](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatStatus.html), which is why I found it appropriate to issue this as a bug.

novatel-applications-engineering, 2021-05-06:
Hi @MCFurry thank you for the additional details. Can you confirm this bug is present in the most recent release? Recently there was work done to address issues similar to what is described.

novatel-applications-engineering, 2021-05-25:
Can you please confirm the issue is present in the most recent release @MCFurry ? Thank you!

MCFurry, 2021-05-25:
Excuse my late answer, but yes, I can confirm the issue is still present!

novatel-applications-engineering, 2022-03-16:
This issue is now fixed as of 4.0.0. If the issues seems to persist, please collect a rosbag with all topics and send it to support.novatel@hexagon.com for further evaluation. Any questions or concerns please let us know.

closed 2022-03-16 by novatel-applications-engineering