novatel/novatel_oem7_driver #20: How to get high-frequency heading value?

zed654, 2021-06-16:
How could I receive high-frequency heading value?
I checked that the default heading data (topic : heading2) has 1 Hz output. But, I wanna get upper than 10 Hz. How could I did it? Is there any command line?

Also, the rostopic of gpdht is not published. Could I receive by using just easy command line?

Thanks !

m4rk0401, 2021-06-18:
I am facing the same problem as you right now. Maybe it is possible to extend the driver with the DUALANTENNAHEADING?

novatel-applications-engineering, 2021-06-23:
Hi @zed654 @m4rk0401. The rate of the HEADING2 log can be adjusted by editing the .yaml files as outlined on our ROS Wiki page here: https://wiki.ros.org/novatel_oem7_driver/configuration

This also steps through adding logs to the files, which will allow you to add the GPHDT log.

If further detail is required, or you feel the above page does not answer your question. Please submit a support ticket to support.novatel@hexagon.com.

zed654, 2021-06-24:
Thanks! I solved the frequency issue by modifying the baud rate from 9600 to 460800 written in ttyUSB.launch.
(On the guideline, the 460800 is recommended if you wanna use IMU log data)

closed 2021-08-12 by novatel-applications-engineering