novatel/novatel_oem7_driver #22: /gps/fix message filled with zeroes

MCFurry, 2021-07-09:
Every now and then we seem to have problems with the ROS driver. The /gps/fix message then does not contain the actual position, but lon/lat fields and covariance fields are filled with zeroes.
If we request the ROS-message /novatel/oem7/bestpos, it does show the valid position here.

What's also interesting, is that when this goes wrong, the field 'altitude' in /gps/fix seems to be populated with exactly the same value as 'undulation' in the bestpos message. 

Whenever things are as expected, the 'altitude' field contains the same value as the 'hgt' field in /novatel/oem7/bestpos.

It almost seems as if the driver chooses to fill the /gps/fix message with other values than those of the bestpos log? Is this a known issue? Or would there be anything else you would need to understand what the issue is?

Thanks in advance!

MCFurry, 2021-07-09:
I recently saw this issue again and was able to capture a screenshot. I then did an `FRESET` on the receiver and restarted the ROS-driver after which the `/gps/fix` message was okay again. These screenshots show exactly the issue:

Good scenario:
[image]

Wrong scenario:
[image]

As can be seen, in the wrong scenario only the `position_covariance_type` and the `altitude` field are populated, where the altitude field in this case actually represents the 'undulation' value with a minus sign?

novatel-applications-engineering, 2021-07-12:
Hi @MCFurry,

Thank you for bringing this to our attention. We're looking into it and will post an update here with our findings for the benefit of the GitHub community

In the meantime, please submit a support request with this information to support.novatel@hexagon.com. This is the best way for our Application Engineers to have an open discussion with you in case we need more information. 

Best regards,
NovAtel Application Engineering

MCFurry, 2021-07-13:
Thanks for your response, I just filed the issue via the support mail as well.
If anything comes up there I will post it here as well.

MCFurry, 2021-07-13:
Just to update here as well:
 - We are using the apt-version of this driver on ros-melodic, hence this seems to be version 2.1.0?
 - While going through the code myself the most logical explanation seems to be that the `inspva` message is interfering? The next time I see this issue again I'll check the output of this ROS-message and report back.

MCFurry, 2021-07-13:
To further update, I captured one of these scenarios with more message outputs:
[image]

MCFurry, 2021-08-17:
For those reading along here, Novatel support has found the issue to be related to picking up the INSPVAX log instead of BESTPOS in my scenario while it actually shouldn't. They are working on a fix which we are testing now.

In the meantime, a workaround is already [in place](https://github.com/novatel/novatel_oem7_driver/blob/7857aab397d27ce5de419146b1b8e1c5e1d89dd5/src/novatel_oem7_driver/src/bestpos_handler.cpp#L754).
By setting the parameter `position_source` to `BESTPOS` you can force the ROS-driver to only use the BESTPOS log for the NavSatFix message. Note though that the rate of this log will be lower than the rate of INSPVAX when available.

novatel-applications-engineering, 2022-01-10:
This fix is in the public repo, both 'master' and '2.3.x'/2.3.2 branches/tags.

closed 2022-01-10 by novatel-applications-engineering