novatel/novatel_oem7_driver #24: Unsupported built in Honeywell IMU

edwardchaos, 2021-08-09:
This current ROS driver version seems to not be compatible with the IMU in my CPT7.

From novatel application suite, when I run "LOG VERSION" command I see that the IMU model is HG4930_AN04
```
LOG VERSION
<OK
[USB2]
<VERSION USB2 0 65.0 FINESTEERING 2170 171730.356 02000800 3681 16482
<     9 
<          GPSCARD "DDDRYNTBEP1" "DMMU20440188T" "OEM7720-1.01" "OM7CR0802RN0000" "OM7BR0100RB0000" "2020/Nov/27" "23:31:48" 
<          OEM7FPGA "" "" "" "OMV070001RN0000" "" "" "" 
<          DEFAULT_CONFIG "" "" "" "SC7CR0802RN0000" "" "2020/Nov/27" "23:32:01" 
<          PACKAGE "" "" "" "SC7PR0802RN0000" "" "2020/Nov/27" "23:31:58" 
<          APPLICATION "" "" "" "SC7AR0802RN0000" "" "2020/Nov/27" "23:31:55" 
<          DB_WWWISO "WWWISO" "0" "" "1.4.0" "" "2021/Jun/01" "10:11:59" 
<          ENCLOSURE "CPT7" "NMZT21290002R" "" "" "" "" "" 
<          WHEELSENSOR "" "" "" "SWS000201RN0000" "" "" "" 
<          IMUCARD "HG4930_AN04" "" "" "" "" "" ""
[USB2]
```

However the closest supported IMU in oem7_supported_imus.yaml is `HG4930_AN01`

Is there a plan on supporting this IMU soon? What can I do for the moment to get full functionality of my device?

novatel-applications-engineering, 2021-08-11:
Hi @edwardchaos this is something that we will update in future releases. The HG4930_AN04 is for the newer generation CPT7 (your model), while the HG4930_AN01 is the previous generation. A work around for this:

The easiest workaround is to set the following ROS parameters in launch file (after including the standard oem7_xx.laucn file)

<param name="/novatel/oem7/supported_imus/68/name" value="Honeywell HG4930 AN04 – 100 Hz" />
<param name="/novatel/oem7/supported_imus/68/rate" value="100" />

<param name="/novatel/oem7/supported_imus/69/name" value="Honeywell HG4930 AN04 – 400 Hz" />
<param name="/novatel/oem7/supported_imus/69/rate" value="400" /> 

 
Alternatively, you could also update the config in the driver, it's just more involved:
Modify: novatel_oem7_driver/src/novatel_oem7_driver/sconfig/oem7_supported_imus.yaml**

add these lines:
"68" : { name: "Honeywell HG4930 AN04 – 100 Hz", rate: "100"}
"69" : { name: "Honeywell HG4930 AN04 – 400 Hz", rate: "400"}

edwardchaos, 2021-08-11:
Thanks, after adding the 2 lines to the list of supported imus, I'm able to run the driver but it not all topics are published as expected.

Specifically, `/gps/imu` is being published but the values are always 0.
[image]

I do see values updated in /novatel/oem7/bestpos however:
[image]

Is this related to the original issue or did I miss something during setup?
I set an INSTRANSLATION for both ANT1 and ANT2 roughly for initial testing. Does it require full SPAN alignment?

novatel-applications-engineering, 2021-08-12:
@edwardchaos Yes, the system will require a full [SPAN configuration](https://docs.novatel.com/OEM7/Content/SPAN_Operation/SPAN_Translations_Rotations.htm?tocpath=Operation%7CSPAN%20Operation%7C_____2) (translations and rotations). 

If you still see further issue, please contact us at support.novatel@hexagon.com and we can debug further.

closed 2021-08-12 by novatel-applications-engineering