novatel/novatel_oem7_driver #26: No imu topic when SPAN is disabled

Timple, 2021-08-16:
**Describe the bug**
No imu topic when SPAN is disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable span
2. Look for imu topic

**Expected behavior**
Imu is still present and active. So could be outputted to ROS.

**Screenshots**

**Environment (please complete the following information):**
 - ROS1 Noetic

**Additional context**
SPAN is not working for us, but for roll and pitch information we would like to use the imu present in the Novatel. Can this information still be made available?

MCFurry, 2021-08-17:
To add to @Timple we are using a SMART7-S receiver in this case (with imu and SPAN integrated).
When disabling SPAN by executing `INSCOMMAND DISABLE`, the imu topic is not published anymore.
Although the rawimu* logs should still be present according to documentation, quote: `Use this command to enable or disable INS. When INS is disabled, no INS position, velocity or attitude is output
(however IMU data is still available).`

edwardribbit, 2021-08-18:
This may be a feature request, the gps/imu topic is actually the corrected imu which relies on SPAN alignment. Is there really a raw imu topic?

MCFurry, 2021-08-18:
Then maybe it is indeed a feature request. 
`gps/imu` is the only imu topic available but we'd really like a raw imu topic as an option!

novatel-applications-engineering, 2021-08-19:
Hi @MCFurry and @edwardribbit.

Thank you for bringing this request to our attention. Can you confirm the request is for:

- Oem7 RAWIMU message output to its own topic (vs. Oem7Raw)

or

- ROS standard imu/data_raw populated per REP 145, based on RAWIMU

Please send those details and the feature request to support.novatel@hexagon.com. We can then track the request with our ROS development team and keep you updated on the progress. 

Best regards,
NovAtel Application Engineering

Timple, 2021-08-20:
For us that would be:
> ROS standard imu/data_raw populated per REP 145, based on RAWIMU

Or even better, `imu/data` because I assume the roll and pitch are known?
Please use the ENU convention as per ROS default. 

Without adhering to REP145 the data cannot easily be processed by other ROS nodes.