novatel/novatel_oem7_driver #27: Add HG4930 AN04 IMU to supported IMUs.

icolwell-as, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
HG4930 AN04 IMU is not in the oem7_supported_imus.yaml list.

**Describe the solution you'd like**
Added to the list if possible.

**Describe alternatives you've considered**
Manually added to list seems to make it work. Would be good to have it already added.

novatel-applications-engineering, 2021-08-19:
This has been added for implementation in future releases. 

The easiest workaround is to set the following ROS parameters in launch file (after including the standard oem7_xx.laucn file)

<param name="/novatel/oem7/supported_imus/68/name" value="Honeywell HG4930 AN04 – 100 Hz" />
<param name="/novatel/oem7/supported_imus/68/rate" value="100" />

<param name="/novatel/oem7/supported_imus/69/name" value="Honeywell HG4930 AN04 – 400 Hz" />
<param name="/novatel/oem7/supported_imus/69/rate" value="400" />
Alternatively, you could also update the config in the driver, it's just more involved:
Modify: novatel_oem7_driver/src/novatel_oem7_driver/sconfig/oem7_supported_imus.yaml**

add these lines:
"68" : { name: "Honeywell HG4930 AN04 – 100 Hz", rate: "100"}
"69" : { name: "Honeywell HG4930 AN04 – 400 Hz", rate: "400"}

novatel-applications-engineering, 2022-01-12:
This has been added to our list of fixes for future releases.

closed 2022-01-12 by novatel-applications-engineering