novatel/novatel_oem7_driver #30: ROS Driver nav_msgs/Odometry message does not update angular velocity

Justin-Ribbit, 2021-10-04:
**Describe the bug**
The ROS driver outputs a nav_msgs/Odometry message under the topic `/novatel/oem7/odom` that contains the filtered pose and velocity estimates. The pose message is complete, however, the twist message only contains the linear velocities. The angular velocities are not being updated.

Would it be possible to update the angular velocities in the nav_msgs/Odometry message? 
 
**To Reproduce**
Steps to reproduce the behavior:
- Run the ROS driver.

**Expected behavior**
A complete nav_msgs/Odometry message containing the full pose and velocity estimates.

**Screenshots**
[image]

**Environment (please complete the following information):**
- OS: Ubuntu 20.04
- Version 2.3.0

oguzhankose, 2022-02-28:
Hi 
Is there any updates on this? I also need angular velocities but i can not get the values.

novatel-applications-engineering, 2022-03-05:
Hi @oguzhankose, let me speak with engineering and see if we can provide an update on this request.

novatel-applications-engineering, 2022-03-07:
Thank you for your input.

Currently, this is an item in our list-to-work-on backlog. We have brought this up again with engineering based on your comment. We will update this issue once we have more update.

vasudevpurohit, 2023-02-02:
Hi!

Was there an update to populate the angular velocities in the odometry topic? Thank you for your time and help!

novatel-applications-engineering, 2023-02-02:
Hi @vasudevpurohit , we have provided a fix for ROS1 here - https://github.com/novatel/novatel_oem7_driver/commit/6799fe00abaa58d818be97afa7d62cda38566ee0

Alternatively, from the NovAtel device - CORRIMUS / IMURATECORRIMUS logs, this outputs the raw IMU data corrected for Gravity/Earth Rate/Estimated Sensor Errors.

https://docs.novatel.com/OEM7/Content/SPAN_Logs/CORRIMUS.htm?Highlight=corrimus

https://docs.novatel.com/OEM7/Content/SPAN_Logs/IMURATECORRIMUS.htm?Highlight=corrimus

We are looking to populate these fields in a future ROS release, but no timetable has been set for this change.