novatel/novatel_oem7_driver #37: Update ros wiki with RTK support infomation

garyedwards, 2021-11-10:
**Is your feature request related to a problem? Please describe.**

The ros wiki documentation includes information on [PPP Correction Services](http://wiki.ros.org/novatel_oem7_driver/configuration#Correction_Services) such as TerraStar but not RTK options like RTK ASSIST. 

**Describe the solution you'd like**

Clarify compatibility with other correction services in the wiki documentation.  If supported include config examples for enabling sharing of data connection with the ROS host computer for RTK correction information.

**Describe alternatives you've considered**

Reviewing the source code and other documentation.

edwardribbit, 2021-11-11:
@garyedwards [this](https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/ANP-074-NTRIP-on-NovAtel-OEM6-and-OEM7/ANP-074-NTRIP-on-NovAtel-OEM6-and-OEM7.pdf) might help

novatel-applications-engineering, 2021-11-18:
Hi @garyedwards, Thank you for your inquiry! Our RTK Assist product that you mention falls under our "TerraStar Corrections Services" banner as it is a PPP based solution used to bridge gaps in RTK service. So the information and requirements in the Wiki applies to it just the same. 

You mentioned: "_enabling sharing of data connection with the ROS host computer for RTK correction information_"

I'm curious if you could elaborate more on this? Do you need RTK data through the driver? Generally RTK corrections are received by the receiver and used in the positioning solution, which is then used to populate the topics of the ROS driver.

closed 2021-12-14 by novatel-applications-engineering