novatel/novatel_oem7_driver #41: imu/data_raw: axis switch within driver. Why?

PhilippSchmaelzle, 2022-01-11:
I am new to this driver and I plan to use the ROS1 Version of it.

With commit f29f11e348b422e3148ac9a5c7efeedf8e36115b the suppport of the `RAWIMUX` is provided! Super cool, thx for this.
But i do not see the reason why the X- and Y axis are switched and negated.

[link to code](https://github.com/novatel/novatel_oem7_driver/blob/f29f11e348b422e3148ac9a5c7efeedf8e36115b/src/novatel_oem7_driver/src/ins_handler.cpp#L275)

**Expected behavior**
As this is a raw signal of the IMU and the log provides this as such (without internal transformation to `RBV` or `USER` frame) I would expect the values to be published in the `IMU` frame (imprinted on the IMU housing)

Or a similar implementation as you did with the `CORRIMU` log where you provide `oem7` as well as `rep 105` conform output. 

For me the remapping of `x = -y` and `y = -x`is not obvious but maybe I just do not see it :smile: 

I hope you can help me with this.

Thanks for this otherwise great driver! As we used our own implementation until now.

Best regards

Timple, 2022-01-11:
The request for the `imu/data_raw` came from us I think. They asked if they should adhere to REP-145: https://github.com/novatel/novatel_oem7_driver/issues/26#issuecomment-902130244

We responded that that should be the case, but somehow it was implemented different :slightly_frowning_face:

PhilippSchmaelzle, 2022-01-11:
Thx @Timple for the hint!
To be consistent within the driver I think the [_RAWIMUX_ data frame]( https://github.com/novatel/novatel_oem7_driver/issues/26#issuecomment-902454706) should also be configurable via `oem7_imu_reference_frame_`. Otherwise users of the _oem7_ frame do not have matching `imu/data_raw` values.

novatel-applications-engineering, 2022-01-13:
Hi @PhilippSchmaelzle @Timple thank you for reporting this issue. We are currently investigating and will provide an update when/if a fix is required. Thank you!

novatel-applications-engineering, 2022-01-18:
Hi @PhilippSchmaelzle @Timple we have resolved this issue in ROS1 branch: https://github.com/novatel/novatel_oem7_driver/commit/8ecee3eb53a754b0780257342308758a65541663

closed 2025-04-09 by novatel-applications-engineering