novatel/novatel_oem7_driver #43: hello,is this driver support Precision Time Protocol （PTP）timesync？

duanqg, 2022-01-17:
**NOTE** 
Feature Requests are not to contain personalized information. Use these forms to submit generalized requests about the open source novatel_oem7_driver. All other inquires should be directed to private communications channels with NovAtel Customer Support.

You may reach NovAtel Customer Support by contacting support.novatel@hexagon.com. Or, you may follow the general Support process explained here: https://docs.novatel.com/OEM7/Content/Front_Matter/Customer_Support.htm

--------------

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

duanqg, 2022-01-22:
thank you！
It is convenient to disclose when this function will be developed？

novatel-applications-engineering, 2022-01-25:
Hello @duanqg, there is currently no timeline for release of this feature. I will update once we have set a formal release date.

JonasLos, 2022-05-05:
Is there an update on when PTP may be supported through this driver directly? If not, is there a work around on getting PTP working without the use of the driver directly?

novatel-applications-engineering, 2022-05-19:
Hi @JonasLos our position on this subject stands the same - there is no current timeline for the release of this feature, however development is ongoing.
Please contact our support email (support.novatel@hexagon.com) for more information on supported time protocols and related schedules.