novatel/novatel_oem7_driver #44: /gps/fix - NavSatFix message altitude in wrong format

PhilippSchmaelzle, 2022-01-21:
**Describe the bug**
The `altitude` within the published [`/gps/fix [NavSatFix]`](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html) message represents mean sea level --> See [Code Link](https://github.com/novatel/novatel_oem7_driver/blob/0b4c3776a5fcf871735e0bce57a3fca599dca93c/src/novatel_oem7_driver/src/bestpos_handler.cpp#L600)

Whereas the `[sensor_msgs/NavSatFix]` expects the height of the WGS84 Ellipsoid.

The altitude within the `INSPVA` is already in the correct format - WGS84 ellipsoid if default is used. The correction of the undulation seams not necessary for me.

**To Reproduce**
- Do a recording with SPAN active.
- compare heights

**Expected behavior**
The published `/gps/fix` message should contain the WGS84 ellipsoid. Not the height of mean sea level.

**Screenshots**
The screenshot shows the three signals as reference.
[image]

**Environment (please complete the following information):**
 - OS: [ubuntu 20.04]
 - ROS1

novatel-applications-engineering, 2022-01-26:
@PhilippSchmaelzle Thank you. We are currently investigating this issue. We will provide an update when we have more information.

novatel-applications-engineering, 2025-04-09:
This was verified and committed here: https://github.com/novatel/novatel_oem7_driver/commit/1b5a49b718e75a9c3aca9ad873bd18fd9cf2a026

closed 2025-04-09 by novatel-applications-engineering