novatel/novatel_oem7_driver #47: ROS buildfarm package update

Timple, 2022-02-08:
Could version 3.0.1 be released to the ROS buildfarm?

Also a 3.0.1 tag seems to be missing from this repository.

I know the version number has only been bumped yesterday, but nobody seems to be watching the [release repository](https://github.com/novatel-gbp/novatel_oem7_driver-release/issues/1)?

novatel-applications-engineering, 2022-02-11:
Thank you for your question. We will work on getting back to your request.

novatel-applications-engineering, 2022-02-25:
We fixed the version tag, changing it from 3.0.1 to 3.0.0. 3.0.0 has been released.

closed 2022-02-25 by novatel-applications-engineering

Timple, 2022-02-28:
But the package.xml still states 3.0.1

Could you please:

- Align the tag with the versions in the package.xml files
- Have the latest tag in the release branch
- Never go back in version numbers. -> This will break a lot of tooling as versions numbers are always expected to increase!

novatel-applications-engineering, 2022-03-05:
Thank you for bringing this issue up. 

Unfortunately, this was a mistake on our end. We are in the process of releasing version 4.0.0, so this should be rectified soon.