novatel/novatel_oem7_driver #50: Requesting support for ROS2 galactic

ajay1606, 2022-02-17:
Novatel OEM7 driver build fails in recent ROS2 distribution (Galactic).

Has any plans to support newer distributions?

Regards
Ajay

novatel-applications-engineering, 2022-02-18:
Hi @ajay1606! Thank you for your question. We currently have plans to support Galactic, but have not had a huge demand thus far so it has not been implemented yet. 

**Please upvote this thread if you are in need of a ROS2 version that supports Galactic.**

ajay1606, 2022-02-19:
Thanks for your quick response, after a few workarounds, have been able to run novate_oem7_driver with ROS2 Galactic distribution without any issues. Confirmed data output without any issues. 

I will report any issues are observed. 

Regards,
Ajay

Minipada, 2022-03-07:
Hey, what fixes did you do? Do you have a fork we could use? Thanks

novatel-applications-engineering, 2022-03-23:
Hello @Minipada, thank you for contacting us. Please see our commit history to ROS1 branch for our most recent changes and fixes applied: https://github.com/novatel/novatel_oem7_driver/commits/ros1

novatel-applications-engineering, 2022-09-09:
Hello @jeremyroy, thank you for your input. We will put in our feature to work on backlog.