novatel/novatel_oem7_driver #51: Missing covariances in imu topic

Timple, 2022-02-24:
**Is your feature request related to a problem? Please describe.**
Thank you for picking up issue #16. The imu message is now usable within the ROS system. For sensor fusion however, the covariance fields needs to be filled.

**Describe the solution you'd like**
Preferably fill the covarience fields with the actual uncertainties.
If these are not known, fill them with the squared value of the stdev from the specsheet. (0.0044 for angular and 0.04 for linear acceleration in the novatel case i think).

**Describe alternatives you've considered**
We now have an 'fix' node inbetween which inserts these values, but this should be fixed at the source.

**Additional context**
We can offer a pull-request if you like.

novatel-applications-engineering, 2022-02-25:
Hi @Timple, thank you for the feedback! I believe this has also been reported as issue #23. We do have this in our queue of items to address and will note there is increased demand for a fix.

Timple, 2022-02-25:
Ah yes, I missed #23.

Would you mind keeping issues open as long as they are issues?

Otherwise one never knows when something is fixed and where to look for open issues.