novatel/novatel_oem7_driver #55: some signals are missing (/imu/data, /imu/data_raw)

kimth1757, 2022-04-07:
Hi, I'm taehan.
I just started using Novatel SPAN solution.

I installed the novatel_oem7_driver and it worked well.
I report some problems, so please suggest a solution.

**1. Issue #1**
    - I can receive /imu/data topics, but all the signals are output as 0.
**2. Issue #2** 
    - I can receive /imu/data_raw topics, but orientations are output as 0.
**3. os**
    - ubuntu 20.04
**4. span solution**
    - pwrpak7 + KVH-1750
**5. std_init_commands.yaml**

# Do not remove these messages -----
- "LOG BESTPOSB ONTIME 0.1" 
- "LOG BESTVELB ONTIME 0.1" 
- "LOG BESTUTMB ONTIME 1"
- "LOG PPPPOSB ONTIME 1"
- "LOG HEADING2B ONNEW"
- "LOG PSRDOP2B ONCHANGED"
- "LOG INSPVASB ONTIME 0.02"
- "LOG INSPVAXB ONTIME 1"
- "LOG CORRIMUSB ONTIME 0.01"
- "LOG INSSTDEVB ONTIME 1"
- "LOG BESTGNSSPOSB ONTIME 0.1"
- "LOG RAWIMUSXB ONNEW"

#--------------------

- "LOG TIMEB ONTIME 1" # Can be used as a 'heartbeat' message.

# Post-Processing Messages; not directly used to generate ROS messages.
- "LOG INSUPDATESTATUSB ONNEW"
- "LOG RAWEPHEMB ONNEW"
- "LOG GLOEPHEMERISB ONNEW"
- "LOG BDSEPHEMERISB ONNEW"
- "LOG GALFNAVEPHEMERISB ONNEW"
- "LOG GALINAVEPHEMERISB ONNEW"
- "LOG RANGEB ONTIME 1"

novatel-applications-engineering, 2022-04-13:
Hello, please send these details to [support.novatel@hexagon.com](mailto:support.novatel@hexagon.com) and we will be happy to look into this for you. Please also include any datasets that show the issue if you have them.

kimth1757, 2022-04-21:
@novatel-applications-engineering Thanks for your reply.
I made 9-DOF imu raw data by combination of /imu/data_raw and /novatel/oem7/inspva messages and it worked.
Best regards,
Taehan.

closed 2022-04-21 by kimth1757