novatel/novatel_oem7_driver #56: Zero-start for odom frame location

pgupta2050, 2022-04-29:
**Problem**
I am using PwrPak7 and this driver for Navigation using ROS Noetic.
I am attempting to utilise the `/novatel/oem7/odom` topic of message type `nav_msgs/Odometry` for odometry information.
This topis by default uses the origin of the current UTM zone as the `odom frame. I need it to use the starting position of the vehicle as the Odom frame.

**Solution I'd like**
The `novatel_span_driver` used to have a param `zero_start` that enabled this functionality. See this functionality here: https://github.com/ros-drivers/novatel_span_driver/blob/4dfd8830b73cf8b4d9ac8d6148b0de3f6990e9b9/novatel_span_driver/launch/example.launch#L47

**Additional context**
If there is already a method with this driver that allows one to do so, please let me know. I could not find it as of now.

Thanks,
Prakhar

novatel-applications-engineering, 2022-05-12:
Hello @pgupta2050, thank you for reaching out to us.

Currently we don't support defining an arbitrary zero for /novatel/oem7/odom. I will put in a new feature request to engineering for you, but we cannot promise on a specific date that this will be implemented.

Please let us know what is your application with our ROS driver and how urgent this is for you.

pgupta2050, 2022-05-13:
Thanks for the response. The application for the ROS driver is to use it for autonomous navigation with an existing autonomous driving stack, which doesn't yet support the Novatel ROS driver. We have remapped the topics output from the driver to interface with the rest of the stack, and the information loop is complete. But the odometry/navigation nodes start off assuming that the `odom` frame location is at the vehicle starting position.
We want to implement this as soon as we can and are already in the process. So, the sooner the better for us :)

osholsonel20, 2022-07-26:
Any updates on this?

novatel-applications-engineering, 2022-07-29:
Hello @osholsonel20, thank you for reaching out to us. This feature is currently in our list of items to work on and we will provide an update here once it is ready.

novatel-applications-engineering, 2023-04-25:
Hi @osholsonel20 and @pgupta2050, thank you for your patience on a response.

This feature is included in ROS2 humble branch. NovAtel has recently kicked off the binary release for humble. For more information, please see: https://github.com/ros/rosdistro/pull/36925 . Soon this will be available for install using Ubuntu's APT package manager, the official date will be published on https://discourse.ros.org/ in a "New Package for Humble" post.

For the time being customer can build from source using the code from humble branch: https://github.com/novatel/novatel_oem7_driver/tree/humble.

Please note that ROS2 driver binaries are currently only available for Jammy (Ubuntu 22.04). Example command to launch ROS2 driver is shown as below:
ros2 launch novatel_oem7_driver oem7_net.launch.py oem7_ip_addr:=192.168.1.10

If you start using this new ROS2 driver, please provide any feedback you may have.

Please let me know if you have any questions.

closed 2023-04-25 by novatel-applications-engineering

novatel-applications-engineering, 2023-04-25:
Link to the specific commit: https://github.com/novatel/novatel_oem7_driver/commit/2af47cf2b039fc21cb2605fbd6cd7e88334b3c83