novatel/novatel_oem7_driver #57: Publishing /TF Frame

osholsonel20, 2022-07-26:
I would like the code to publish the /tf frames of the odom frame.

Example here:
http://docs.ros.org/en/melodic/api/novatel_span_driver/html/publisher_8py_source.html

Line 95 - Publish tf

novatel-applications-engineering, 2022-07-29:
Hello @osholsonel20, thank you for reaching out to us. 

We have started a conversation with engineering regarding your request. We will update you once we learn more.

novatel-applications-engineering, 2022-09-09:
Hello @osholsonel20, thank you for your patience on a response. This feature request has been added to our list of features to be added.

novatel-applications-engineering, 2023-04-25:
Hi @osholsonel20, thank you for your patience on a response.

This feature has been implemented in ROS2 humble driver. NovAtel has recently kicked off the binary release for humble. For more information, please see: https://github.com/ros/rosdistro/pull/36925 . Soon this will be available for install using Ubuntu's APT package manager, the official date will be published on https://discourse.ros.org/ in a "New Package for Humble" post.

For the time being customer can build from source using the code from humble branch: https://github.com/novatel/novatel_oem7_driver/tree/humble.

Please note that ROS2 driver binaries are currently only available for Jammy (Ubuntu 22.04). Example command to launch ROS2 driver is shown as below:
ros2 launch novatel_oem7_driver oem7_net.launch.py oem7_ip_addr:=192.168.1.10

If you start using this new ROS2 driver, please provide any feedback you may have.

Please let me know if you have any questions.

closed 2023-04-25 by novatel-applications-engineering