novatel/novatel_oem7_driver #58: Streaming RTCM corrections to the device from ROS

garyedwards, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
Support sending streaming RTCM corrections to the device from a ROS topic.

ROS [NTRIP Clients](https://github.com/LORD-MicroStrain/ntrip_client) already exist for receiving RTCM corrections from NTRIP servers and publishing them to a ROS topic.

It would be good if this driver could support sending these RTCM corrections to the device as implemented in [other](https://github.com/LORD-MicroStrain/microstrain_inertial), [drivers](https://xsenstechnologies.force.com/knowledgebase/s/article/Using-an-NTRIP-client-with-the-Xsens-ROS-driver?language=en_US).

This allows for re-using the data connection from the ROS computer.

Support any NTRIP servers that can have a client writing to a ROS topic.

Share NTRIP server connection and RTCM corrections code with other RTK drivers.

**Describe the solution you'd like**
Streaming RTCM corrections to the device from a [nmea_msgs/Sentence](http://docs.ros.org/en/api/nmea_msgs/html/msg/Sentence.html) ROS topic.

**Describe alternatives you've considered**
Using the native Novatel RTCM corrections setup.

**Additional context**
N/A

novatel-applications-engineering, 2022-10-21:
Hello @garyedwards, thank you for reaching out to us. I have brought this up with our engineering team and we have decided to put in a new feature request for you. 

Currently, we await product management's decision. Once a decision has been made we will update you.

novatel-applications-engineering, 2025-01-03:
@mbed92 @garyedwards Hi, thank you for your patience on a response. This has remained on our list of ROS backlog, but we are currently re-discussing internally the viability and possibility of implementing such a feature. Would you be able to provide more insights on this feature request and its use case?

Note that NovAtel receiver, such as PwrPak7, has a built-in NTRIP client to receive RTCM corrections over NTRIP via ethernet (assuming that the receiver has direct access to the internet). For more detail, please see https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/APN-074-NTRIP-on-NovAtel-OEM6-OEM7/APN-074-NTRIP-on-NovAtel-OEM6-OEM7.pdf

Does this NTRIP client feature solve your issue?