novatel/novatel_oem7_driver #59: Document best practice for overriding default params

garyedwards, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
Document best practice for overriding default params.

The `oem7_ext_params_file` arg in the `std_driver_config.xml` file seemed like a good solution to this but has been [removed](https://github.com/novatel/novatel_oem7_driver/commit/e767758fdc453074e074144b7949490431a04641).

At the moment the only solution I can think of is to maintain local copy of the whole `std_driver_config.xml` file.  It does not seem to change much so maybe this is fine and what is expected?

**Describe the solution you'd like**
Confirm the above local `std_driver_config.xml` is the expected way to update parameters.

Exposing more args in the upstream `std_driver_config.xml` would also be a nice option.

**Describe alternatives you've considered**
As above.

**Additional context**
This is to mainly support [`receiver_init_commands`](http://wiki.ros.org/novatel_oem7_driver/configuration#Customizing_Logging_Configuration) customisation which I would imagine most users need to do.

novatel-applications-engineering, 2022-08-25:
Hi @garyedwards, thank you for reaching out to us. We have received your request and will initiate a discussion with engineering.

We will let you know when an update is available.

novatel-applications-engineering, 2022-09-09:
Hi @garyedwards, would you be looking at using this feature in ROS1 only? or is this also requested for ROS2?

garyedwards, 2022-09-21:
Both really but we are using ROS1 at the moment.

novatel-applications-engineering, 2022-09-26:
Hi @garyedwards, thank you for your input. I will pass this information to our product management and engineering.

novatel-applications-engineering, 2022-10-19:
Hi @garyedwards 
Thank you for your patience on a response. 
After discussing with engineering, this is a part of best practices for using ROS. The extended parameters should be a part of the launch file not the xml file. This removes any irrelevant hard coded instructions that can impact reusability.
Nevertheless, you still have the ability to add them on your local drive for your convenience.
Please let us know if you have any questions.

closed 2025-04-09 by novatel-applications-engineering