novatel/novatel_oem7_driver #60: driver crash issue,  error message is  "Bond broken,exiting" "Oem7Receiver: read error: 9; endpoint open: 1 errors/max: 1/7"

Maxinjun, 2022-08-24:
novatel ros driver crash, the error log as below
"Oem7Receiver: read error: 9; endpoint open: 1 errors/max: 1/7"

By the way , it recovered fine after I launch the driver program again, so I think it may be the ros driver bug.

here is the terminator log:
===========================================
localization status is ok ===================56
[ INFO] [1661243890.926129756]: Log Statistics:
[ INFO] [1661243890.926194785]: Logs: 4830000; unknown: 52; discarded: 0
[ INFO] [1661243890.926221762]: Log[RAWEPHEM](41):1510
[ INFO] [1661243890.926235669]: Log[BESTPOS](42):104441
[ INFO] [1661243890.926249654]: Log[RANGE](43):10444
[ INFO] [1661243890.926262951]: Log[RXSTATUS](93):643
[ INFO] [1661243890.926275570]: Log[BESTVEL](99):104442
[ INFO] [1661243890.926288631]: Log[TIME](101):10443
[ INFO] [1661243890.926305054]: Log[INSPVAS](508):1044484
[ INFO] [1661243890.926323815]: Log[GLOEPHEMERIS](723):1327
[ INFO] [1661243890.926339222]: Log[BESTUTM](726):10444
[ INFO] [1661243890.926356457]: Log[PSRDOP2](1163):8231
[ INFO] [1661243890.926373922]: Log[HEADING2](1335):843
[ INFO] [1661243890.926401493]: Log[BESTGNSSPOS](1429):104441
[ INFO] [1661243890.926422953]: Log[RAWIMUSX](1462):1305670
[ INFO] [1661243890.926446075]: Log[INSPVAX](1465):1044434
[ INFO] [1661243890.926465998]: Log[PPPPOS](1538):10443
[ INFO] [1661243890.926486461]: Log[BDSEPHEMERIS](1696):3519
[ INFO] [1661243890.926512787]: Log[TERRASTARINFO](1719):36
[ INFO] [1661243890.926534943]: Log[TERRASTARSTATUS](1729):36
[ INFO] [1661243890.926558297]: Log[INSUPDATESTATUS](1825):9243
[ INFO] [1661243890.926578993]: Log[INSCONFIG](1945):37
[ INFO] [1661243890.926600011]: Log[INSSTDEV](2051):10444
[ INFO] [1661243890.926621025]: Log[CORRIMUS](2264):1044445
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
[ INFO] [1661243909.406845641]: Bond broken, exiting
[ INFO] [1661243909.406869981]: Bond broken, exiting
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
[novatel/oem7/receivers/main-2] process has finished cleanly
log file: /home/neousys/.ros/log/86b0b510-22a6-11ed-b5f7-78d0042dd27d/novatel-oem7-receivers-main-2*.log
[novatel/oem7/receivers/main/config-3] process has finished cleanly
log file: /home/neousys/.ros/log/86b0b510-22a6-11ed-b5f7-78d0042dd27d/novatel-oem7-receivers-main-config-3*.log
localization status is ok ===================56
localization status is ok ===================56
localization status is ok ===================56
[ INFO] [1661243912.612160141]: Log Statistics:
[ INFO] [1661243912.612224046]: Logs: 4840000; unknown: 52; discarded: 0
[ INFO] [1661243912.612260117]: Log[RAWEPHEM](41):1510
[ INFO] [1661243912.612287102]: Log[BESTPOS](42):104658
[ INFO] [1661243912.612308142]: Log[RANGE](43):10466
[ INFO] [1661243912.612329005]: Log[RXSTATUS](93):644
[ INFO] [1661243912.612353055]: Log[BESTVEL](99):104659
[ INFO] [1661243912.612373710]: Log[TIME](101):10465
[ INFO] [1661243912.612396829]: Log[INSPVAS](508):1046653
[ INFO] [1661243912.612418271]: Log[GLOEPHEMERIS](723):1327
[ INFO] [1661243912.612439306]: Log[BESTUTM](726):10466
[ INFO] [1661243912.612460102]: Log[PSRDOP2](1163):8231
[ INFO] [1661243912.612482975]: Log[HEADING2](1335):843
[ INFO] [1661243912.612504625]: Log[BESTGNSSPOS](1429):104658
[ INFO] [1661243912.612525689]: Log[RAWIMUSX](1462):1308380
[ INFO] [1661243912.612547100]: Log[INSPVAX](1465):1046603
[ INFO] [1661243912.612568000]: Log[PPPPOS](1538):10465
[ INFO] [1661243912.612589326]: Log[BDSEPHEMERIS](1696):3519
[ INFO] [1661243912.612610711]: Log[TERRASTARINFO](1719):36
[ INFO] [1661243912.612632347]: Log[TERRASTARSTATUS](1729):36
[ INFO] [1661243912.612653894]: Log[INSUPDATESTATUS](1825):9265
[ INFO] [1661243912.612675145]: Log[INSCONFIG](1945):37
[ INFO] [1661243912.612696721]: Log[INSSTDEV](2051):10466
[ INFO] [1661243912.612720116]: Log[CORRIMUS](2264):1046613
[ERROR] [1661243913.321724023]: Oem7Receiver: read error: 9; endpoint open: 1 errors/max: 1/7
[ERROR] [1661243913.321764178]: Oem7Receiver: close error: 9
[novatel/oem7/driver-1] process has died [pid 4364, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=driver __log:=/home/neousys/.ros/log/86b0b510-22a6-11ed-b5f7-78d0042dd27d/novatel-oem7-driver-1.log].
log file: /home/neousys/.ros/log/86b0b510-22a6-11ed-b5f7-78d0042dd27d/novatel-oem7-driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
neousys@neousys-Nuvo-8208GC-Series:~/bin$ 
=============================================================

novatel-applications-engineering, 2022-10-24:
Hello @Maxinjun 

Thank you for reaching out to us and reporting this issue.
Can you provide more information about how to reproduce this issue (setup details, ROS driver version, etc...)?

novatel-applications-engineering, 2022-11-18:
Hello @Maxinjun 
I am just following up on this case.
Can you provide more information about how to reproduce this issue (setup details, ROS driver version, etc...)?

novatel-applications-engineering, 2023-03-16:
Hello, I believe this should be resolved with the latest update to the ROS1 branch and binaries to compile novatel_edie at runtime instead of using an out of date pre-compiled binary. I will close this issue for now but please re-open or reach out if you are still encountering errors.

closed 2023-03-16 by novatel-applications-engineering