novatel/novatel_oem7_driver #61: IMU Data does not conform to REP105

zdjeffri, 2022-10-03:
**Describe the bug**
Despite #16, raw imu data still does not conform to REP105. This is important for certain applications in robot SLAM.
I understand raw data is only reference to the sensor body (and this forward, right, up) is arbitrary and the INSSetRotation commands do not apply.
The raw data can still easily be made ROS Compliant by remapping the gyros. 

**To Reproduce**
In RViz, you can see how the raw data is rotated 90* from the composite data.

**Expected behavior**
Raw data should match (in orientation) the composite data. If the module is mounted as recommended, i.e. wire on back, pitching the vehicle forward should result in a backward acceleration vector viewing raw data.

**Screenshots**
Actual: 
[image]
Expected: 
[image]

**Additional context**
This is a simple fix. I will issue a pull request containing an additional publisher that has remapped the gyro/accel as needed for REP105

novatel-applications-engineering, 2022-10-04:
Hi @zdjeffri, thank you for reaching out to us. We will work on reproducing this issue on our side and let engineering team know. We will update you once we learn more.

novatel-applications-engineering, 2022-10-11:
Hi @zdjeffri, thank you for submitting a pull request (PR). After reviewing this PR. We have decided not to merge this into our current code base. 

For OEM7, the raw IMU data is with respect to the IMU installation and the corresponding direction of the axis. Users have the ability to mount the IMU many different ways; for example, X axis forward, Z axis to the right, and Y axis pointing up. When the IMU is mounted in such position, the gyro information will need to change accordingly to be conformed to REP105. The current PR does not address these different IMU installation cases.

The fix provided to address [#16](https://github.com/novatel/novatel_oem7_driver/issues/16) is [Issue 16: Make IMU message REP105 compliant.](https://github.com/novatel/novatel_oem7_driver/commit/cf3127689d6104cd75ed0718086d1e81f3a78fdb); and it publishes the [CORRIMUDATA](https://docs.novatel.com/OEM7/Content/SPAN_Logs/CORRIMUDATA.htm), which is already in the vehicle frame and rotates it to conform with REP105. This is published in topic /gps/imu.

CORRIMUDATA is the IMU data corrected for gravity, earth's rotation, and estimated sensor error.

Would this solution work for you?

Please let us know.

closed 2025-04-09 by novatel-applications-engineering