novatel/novatel_oem7_driver #63: Why does inspva under ROS use ros::time for timestamp even though my device is connected to GPS?

leslieburke, 2022-10-27:
If I change the system time to a wrong time, the time of the topic in ROS will also be wrong and will not be corrected by GPS.

novatel-applications-engineering, 2022-11-01:
Hi @leslieburke, thank you for reaching out to us. We will work on reproducing this issue on our side and let engineering team know. We will update you once we learn more.

novatel-applications-engineering, 2022-12-19:
Hi @leslieburke 

Please be advised that generally ROS interacts with multiple sensors and systems, which are not necessarily connected to GNSS. Hence ROS doesn't have a notion of 'GPS time', only system time. 

Nevertheless, UTC time can be calculated and logged in the rostopic /novatel/oem7/time 

For more reference about the NovAtel Time logs, you can refer to:
https://docs.novatel.com/OEM7/Content/Logs/TIME.htm

Another similar inquiry was posted on ROS answers:
https://answers.ros.org/question/311325/properly-timestamping-gnss-data/

closed 2022-12-20 by novatel-applications-engineering