novatel/novatel_oem7_driver #64: Get source code of liboem7decoder.a which in arm64v8 and amd64

liurunhan000, 2022-11-08:
**NOTE** 
Feature Requests are not to contain personalized information. Use these forms to submit generalized requests about the open source novatel_oem7_driver. All other inquires should be directed to private communications channels with NovAtel Customer Support.

You may reach NovAtel Customer Support by contacting support.novatel@hexagon.com. Or, you may follow the general Support process explained here: https://docs.novatel.com/OEM7/Content/Front_Matter/Customer_Support.htm

--------------

**Is your feature request related to a problem? Please describe.**
I am working on transplanting novatel ome7 driver to another system(QNX) but unfortunately this system doesn's support libc.so.6 so i wonder if there's a way to get source code of liboem7decoder.a  that i can do some modification

**Describe the solution you'd like**
Get source code of liboem7decoder.a which in arm64v8 and amd64

**Additional context**
here is the compile bug
```
novatel_oem7/bin/arm64v8/liboem7decoder.a(oem7_message_decoder_lib.o):function boost::shared_ptr<StandardDecoder>::operator->() const: error: undefined reference to '__assert_fail'
novatel_oem7/bin/arm64v8/liboem7decoder.a(novatelparser.o):function unsigned long __gnu_cxx::__stoa<unsigned long, unsigned long, char, int>(unsigned long (*)(char const*, char**, int), char const*, char const*, unsigned long*, int)::_Save_errno::_Save_errno(): error: undefined reference to '__errno_location'
```

novatel-applications-engineering, 2022-11-18:
Hello @liurunhan000 
Your request has been raised and we are working on it.
The source for the liboem7decoder.a is currently available in the branch ros2-dev. You can access the library from there (oem7_message_decoder.lib.cpp/hpp).
We are currently working on importing the library and testing the whole software, once this is done the source code for the libraries will be pushed to the main. Due to the resource constrain on our engineering team, the timeline for this is TBD.

liurunhan000, 2022-11-24:
thank you very much!

closed 2023-01-17 by liurunhan000

novatel-applications-engineering, 2023-02-06:
Hello @liurunhan000 
The source code for the libraries are now available in the branch
https://github.com/novatel/novatel_oem7_driver/tree/ROS1_add_src_1
We have compiled and tested it on our side, it should be working fine.

liurunhan000, 2023-02-22:
thank you so much!