novatel/novatel_oem7_driver #65: [novatel_oem7_driver] Ros2 humbel build failures

starmin95, 2023-01-05:
hi there

is it correct that novatel_osm7_driver is supported by ros2 humble??
I tried to build(./build.sh -f), but an error occurred as shown in the attached picture.

how can i solve the problem?

Environment
 - OS: Ubuntu
 - Version : 22.04
-  ROS2 Version : ROS2 Humbel
-  used novatel_osm7_driver git branch : humble

[image]

novatel-applications-engineering, 2023-01-11:
Hello @starmin95 
We have tried to reproduce the issue on our side and have been unable to.
Can you please provide us with 'LastTest.log' mentioned in the console output below? This will help us further investigate the issue.
Please send the log attached to an email to: support.novatel@hexagon.com. As well, please mention GitHub issue#65 in the title of your email.

llu0120, 2023-02-17:
Hi @starmin95 , did you figure out how to solve this problem? Look forward to your reply, thank you.

novatel-applications-engineering, 2023-02-28:
Hi @llu0120
Have you encountered a similar issue? 
If so, can you please provide us with 'LastTest.log'? This will help us investigate the issue.

MortezaAliyari, 2024-02-23:
@novatel-applications-engineering 
I have the same issue. is there any solution?!

closed 2025-04-09 by novatel-applications-engineering