novatel/novatel_oem7_driver #66: Segfault in libnovatel_oem7_driver.so sometimes on startup

MCFurry, 2023-02-01:
On some machines I sometimes experience a segfault of the driver nodelet when starting this driver under ROS-noetic (v4.0.0)

A coredump I was able to get out is uploaded here:
https://mcfurry.stackstorage.com/s/pF3ygkQJG08cqN01

This is what I recovered from the systemlog:

```Feb 01 19:20:43 6a6adfa audit[197500]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=197500 comm="nodelet" exe="/opt/ros/noetic/lib/nodelet/nodelet" sig=11 res=1
Feb 01 19:20:43 6a6adfa kernel: show_signal_msg: 10 callbacks suppressed
Feb 01 19:20:43 6a6adfa kernel: nodelet[199181]: segfault at 58 ip 00007f8eec715e12 sp 00007f8ed6ffc430 error 4 in libnovatel_oem7_driver.so[7f8eec672000+254000]
Feb 01 19:20:43 6a6adfa kernel: Code: 48 89 51 38 8b 45 e0 89 41 40 48 8b 45 b8 8b 40 34 85 c0 74 29 48 8b 45 b8 48 8b 10 48 8b 45 b8 48 8b 00 48 8b 00 48 83 c0 58 <48> 8b 00 48 89 d7 ff d0 83 f8 01 74 07 b8 01 00 00 00 eb 05 b8 00
Feb 01 19:20:43 6a6adfa kernel: audit: type=1701 audit(1675279243.016:206): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=197500 comm="nodelet" exe="/opt/ros/noetic/lib/nodelet/nodelet" sig=11 res=1
```

A fresh restart of the driver seems to always recover and the nodelets keep on working from then on. Apparently it seems to be a startup issue?

novatel-applications-engineering, 2023-02-01:
Hello @MCFurry,

Thank you for the information. I will take a look at the core dump and see if it gives any clues. In the meantime, I was hoping you could provide me with some information about your setup / environment in hopes that we can try to reproduce the problem on our end.

1. Are you building our ROS driver yourself, or using a pre-compiled binary?
2. What version of Ubuntu are you running?
3. Does this happen every time you first start the driver on a machine?
4. What kind of receiver are you interfacing with? If at all possible, do you have or could you provide raw receiver data (from the serial port or however you are interfacing with the receiver) from start-up when this condition occurs? This would allow us to check whether or not there is something in the receiver data causing the fault to occur. 

Please send us an email and any other logs attached to: [support.novatel@hexagon.com](mailto:support.novatel@hexagon.com). As well, please mention GitHub issue #66 in the title of your email.

MCFurry, 2023-02-02:
I'll also pop this by email, but hereby the requested information:

1. We're using the pre-compiled binary from apt (ros-noetic-novatel-oem7-driver latest version: 4.0.0)
2. Standard Ubuntu 20.04 server edition
3. We mostly encounter the issue when starting the driver as part of a ROS-stack directly after boot
4. We're interfacing with a SMART7S receiver, I wouldn't be able to give you raw serial data on a short term, so hopefully the coredump points into a direction already?

novatel-applications-engineering, 2023-02-02:
Thank you,

I was able to find the problem using your core debug file and we are working on fixing it. I will let you know when we have an update that resolves this.

MCFurry, 2023-02-02:
Awesome, thanks for the quick investigation! Looking forward to the fix!

novatel-applications-engineering, 2023-02-27:
Hello @MCFurry, the fix is now published to the ROS1 branch. There will be a binary release of the updated code coming soon. ROS1 now uses the up to date external NovAtel EDIE repository instead of pre-compiled binaries.

closed 2023-02-27 by novatel-applications-engineering