novatel/novatel_oem7_driver #67: Build from source for ROS2 Humble on Ubuntu 22.04 (aarch64) failing

prennich, 2023-02-19:
**Describe the bug**
The build.sh script is failing to produce .deb files for installation.

**To Reproduce**
Here are the steps I've taken:

$ git clone https://github.com/novatel/novatel_oem7_driver.git
$ cd novatel_oem7_driver
$ git switch humble
$ rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully
$ export ROS_DISTRO=humble
$ source /opt/ros/humble/setup.bash
$ ./build.sh -f

**Expected behavior**
Was expecting .debs to be created.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04 on aarch64 (RaspberryPi 4)
 - ROS Humble installed and functioning.

**Additional context**
Below is the command line output after executing ./build.sh -f:

Starting >>> novatel_oem7_msgs
[Processing: novatel_oem7_msgs]                            
[Processing: novatel_oem7_msgs]
[Processing: novatel_oem7_msgs]         
[Processing: novatel_oem7_msgs]                       
[Processing: novatel_oem7_msgs]                               
[Processing: novatel_oem7_msgs]                             
Finished <<< novatel_oem7_msgs [3min 20s]       
Starting >>> novatel_oem7_driver
[Processing: novatel_oem7_driver]           
[Processing: novatel_oem7_driver]                     
[Processing: novatel_oem7_driver]         
[Processing: novatel_oem7_driver]                 
[Processing: novatel_oem7_driver]                  
[Processing: novatel_oem7_driver]                         
[Processing: novatel_oem7_driver]                           
[Processing: novatel_oem7_driver]                      
[Processing: novatel_oem7_driver]                         
[Processing: novatel_oem7_driver]                         
[Processing: novatel_oem7_driver]                           
[Processing: novatel_oem7_driver]                  
--- stderr: novatel_oem7_driver                            
CMake Warning at /opt/ros/humble/share/gps_tools/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'gps_tools' exports library 'gps_tools' which couldn't be found
Call Stack (most recent call first):
  /opt/ros/humble/share/gps_tools/cmake/gps_toolsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

Linking to EDIE at: '/home/aloft/novatel/novatel_oem7_driver/build/novatel_oem7_driver'
Cloning into 'novatel_oem7_decoder'...
HEAD is now at ebb311d -Wno-format-security
HEAD is now at ebb311d -Wno-format-security
CMAKE Host :  Linux
CMAKE_SYSTEM_PROCESSOR:aarch64
CMAKE_LIB_SHARED:
ARCH is not Set, default to x64
ARCH is x64 & Building for 64 bit
CMAKE_C_FLAGS_DEBUG:    -g
CMAKE_CXX_FLAGS_DEBUG:  -g -std=gnu++11 -fPIC -D_LINUX_
CMAKE_C_FLAGS_RELEASE:  -s -O3 -DNDEBUG
CMAKE_CXX_FLAGS_RELEASE:-g -Wno-write-strings -Wno-format -Wno-format-security -Wno-overflow -std=gnu++11 -fPIC -D_LINUX_ -DNDEBUG -s
CMake Deprecation Warning at lib/googletest/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

[... 100 lines cut ...]
Expanding 'debian/rules.em' -> 'debian/rules'
dh binary -v --buildsystem=cmake --builddirectory=.obj-aarch64-linux-gnu
make: dh: No such file or directory
make: *** [debian/rules:27: binary] Error 127
[sudo] password for aloft: Sorry, try again.
[sudo] password for aloft: 
sudo: no password was provided
sudo: 1 incorrect password attempt
[sudo] password for aloft: Sorry, try again.
[sudo] password for aloft: 
sudo: no password was provided
sudo: 1 incorrect password attempt
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning:  is an invalid version and will not be supported in a future release
  warnings.warn(
ROS Distro index file associate with commit '91349fa4751c7a5b62b1129220148a11c693694f'
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/91349fa4751c7a5b62b1129220148a11c693694f/index-v4.yaml'
==> Generating debs for ubuntu:jammy for package(s) ['novatel_oem7_driver']
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning:  is an invalid version and will not be supported in a future release
  warnings.warn(
Could not resolve rosdep key 'novatel_oem7_msgs'
Try to resolve the problem with rosdep and then continue.
Continue [Y/n]? n
GeneratorError: Error running generator: Failed to resolve rosdep key 'novatel_oem7_msgs', aborting.
/usr/bin/fakeroot: 175: debian/rules: not found
mv: cannot stat 'src/*.deb': No such file or directory

novatel-applications-engineering, 2023-02-23:
Thank you for reaching out to us @prennich. We will work on reproducing this issue locally and provide a solution afterwards. I will update you once I learn more.

novatel-applications-engineering, 2023-03-02:
Hi @prennich, the current solution is to change the 'rosdep.yaml' to (we will be pushing this change to humble branch, but for now you will need to do this locally):

novatel_oem7_msgs: 
    ubuntu: 
        focal:  [ros-foxy-novatel-oem7-msgs]
        jammy: [ros-humble-novatel-oem7-msgs]

Save the file, and then start the docker environment using the instruction shown in https://github.com/novatel/novatel_oem7_driver/tree/humble with command: './docker/run.sh -r amd64 humble'

Once the docker environment is built, then run: './build.sh -f'

prennich, 2023-03-02:
Thank you. I will test this solution and report back.

novatel-applications-engineering, 2023-03-02:
Change committed to humble branch.

closed 2023-03-02 by novatel-applications-engineering