novatel/novatel_oem7_driver #68: Support for dual pwrpak use on single ROS network

frenklakAloft, 2023-03-13:
Hello, for my system, I would like to operate two PwrPaks simultaneously, receiving the same variables e.g. INSPVAX over a single ROS network from each novatel unit. However, the novatel_oem7_driver sets ROS fields as global variables, rather than prefix-ing them with the namespace from which the driver is called. This leads to a conflict in the definitions of the two variable streams. Is there some adjustment that can be made, whether to the novatel ROS package itself, or in the launch file definitions, such that the variable names can be altered to avoid conflict, such as by prefix-ing with namespace, or by manually renaming (e.g. INSPVAX_1 vs. INSPVAX_2) ? Thank you

novatel-applications-engineering, 2023-04-06:
Hello freklakAloft,

Which specific version and branch of ROS are you using?

For ROS1, the functionality you're looking for was baked into branch 4.0.1: https://github.com/novatel/novatel_oem7_driver/tree/4.0.1
An example of launching two instances of the ROS driver for 2 receivers is shown in this launch file here: https://github.com/novatel/novatel_oem7_driver/blob/4.0.1/src/novatel_oem7_driver/launch/oem7_net_2_ns.launch

Please note: you will have to build the driver binary files from the source code in the 4.0.1 branch to use this functionality.

If you are using ROS2, please let us know the exact build you are using, so we can look into how it should be done for your specific ROS2 build.

frenklakAloft, 2023-05-01:
Thank you -- I was able to get branch 4.0.1 working with my ROS1 installation (noetic) and two simultaneous driver instances, so this was a great help. There is a remaining bug relating to the USB/tty usage on that branch, but I'll follow up with a new thread.

closed 2023-05-01 by frenklakAloft