novatel/novatel_oem7_driver #72: Transformation between gps frame and imu_link

chinitaberrio, 2023-05-04:
The ROS driver publishes `/gps/fix` data in the `gps` frame and `gps/imu` data in the `imu_link` frame; however, there is an absence of any topic between these two reference frames (tf_static). Should this transformation come from the INSCONFIG?

novatel-applications-engineering, 2023-05-19:
Hi @chinitaberrio, thank you for reaching out to us. 

I have passed your request to engineering. We will work on providing a set of instructions for you to work with tf_static. I will update you once I learn more.

novatel-applications-engineering, 2023-06-07:
Hi @chinitaberrio, will you be using the SPAN INS solution? or will you be performing your own GNSS and IMU solution fusion? I would like to understand your application first before providing a recommendation.

/gps/fix topic can take position source from BESTPOS (INS position solution translated to the antenna phase center if position type has INS in it) and INSPVAS (INS position solution at default IMU center of navigation location unless SETINSTRANSLATION USER command is used). Reference: https://wiki.ros.org/novatel_oem7_driver/ros1_topic_mapping

User can select which position source that the ROS driver uses, by defining the position source parameter to 'BESTPOS' or 'INSPVAS'. In ROS1, use 'position_source' parameter in launch file. In ROS2 (humble), use 'oem7_position_source' parameter in launch file.

INSCONFIG provides the lever arm translations and rotations. There could be multiple of them depending on user application. At minimum, in a correctly configured system, you will see at least the lever arm from IMU center of navigation to antenna phase center (this can be in the vehicle frame or the IMU body frame, depending on the command used) and RBV (rotation from IMU body frame to vehicle frame).

Frame definition can be found at: https://docs.novatel.com/OEM7/Content/SPAN_Operation/Definition_Reference_Frames.htm

We can also set up for a brief call if this does not provide clear guidance. Please let me know.

chinitaberrio, 2023-06-30:
Hi @novatel-applications-engineering 

We're using the SPAN INS solution. We are still a bit confused about the /gps/fix frame's context. In our current configuration, we have not set up the position_source. Therefore, based on the driver's information:
[position_source = "BESTPOS or INSPVAS based on quality"](https://protect-au.mimecast.com/s/lr3bC81V0PTXx31OZT4BIYR?domain=github.com)

According to the information provided, when the position_source is set to BESTPOS, the frame for /gps/fix should be 'gps'. However, when INSPVAS is used in GPS-denied areas, the frame should be 'imu'. However, we examined a recorded rosbag that captured an RTK signal at the beginning and entering a tunnel afterwards. We were expecting to observe a switch from the 'gps' frame to the 'imu' frame (in the header of gps/fix)  when entering the tunnel, and it doesn’t happen. It is confusing when the driver does not provide the static transformation between all the frames, especially when it should have this information already. Furthermore, it does not make much sense to have a topic switching between different reference frames when the ROS driver already has knowledge of the transformations between them.

I'll be more than happy to jump on a video call!

novatel-applications-engineering, 2025-04-09:
There are three way of ensuring position solution output location doesn't change:

- (recommended) configure SETINSTRANSLATION USER in std_init_commands.yaml file with primary lever arm value, this will translate the INS position solution from IMU center of navigation to antenna phase center, when the position source changes from BESTPOS to INSPVAS, you will not observe an offset in position solution.
- change position source to BESTPOS, maximum output rate of 20Hz
- change position source to INSPVAS, maximum output rate of 200Hz

INS configuration needed for your system to operate properly:

- Primary lever arm is configured with SETINSTRANSLATION ANT1 command
- IMU frame is known by the INS system itself, with rotation from IMU body frame to vehicle frame, vehicle frame is also know. This is configured with SETINSROTATION RBV command.
- If you have a secondary antenna, for INS to use dual antenna heading to help INS to achieve initialization faster, at least one of the conditions below must be met:
- Secondary lever arm is configured with SETINSTRANSLATION ANT2 command
- SETINSROTATION ALIGN is configured manually or by calibration

closed 2025-04-09 by novatel-applications-engineering