novatel/novatel_oem7_driver #74: Inconsistent publishing Frequency

JonasLos, 2023-06-08:
**Describe the bug**
/novatel/oem7/odom and /gps/gps header show that the topics are published at an inconsistent frequency. On average, we are seeing about 0.05 seconds of no message being published once per second.
This happen at several frequencies (setting the publishing rate to 50 Hz and 100 Hz makes no difference in this delay) 

**To Reproduce**
Steps to reproduce the behavior:
We have tried several different approaches. We used tcp and udp to connect to the device
roslaunch novatel_oem7_driver oem7_net.launch oem7_ip_addr:=192.168.100.150 oem7_port:=3002 oem7_if:=Oem7ReceiverUdp
roslaunch novatel_oem7_driver oem7_net.launch oem7_ip_addr:=192.168.100.150 oem7_port:=3001
without change in the behavior. 
We also recorded rosbags with different tags - such as --udp and --tcpnodelay with no mentionable difference. It does seem that sometimes the delay is worse, up to 0.1 seconds or more. 

**Expected behavior**
I would expect the messages to be published at a consistent rate. 

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04, ROS melodic
 - Latest Firmware on the PwrPak7D

**Additional context**
additional rosbags were recorded with different settings - seemingly making no difference what-so-ever. Link to a drive with [rosbags:](https://drive.google.com/drive/folders/1WkXFOm2Pfq8-EREzt27iXP3XpO1ihnI7?usp=sharing)

Timple, 2023-06-15:
Just to chime in, we have the same issue.

novatel-applications-engineering, 2023-06-15:
Hi @Timple and @JonasLos, thank you for reaching out to us.

I am currently investigating this issue locally, and I will provide an update once I learn more.

novatel-applications-engineering, 2023-06-20:
Hi @Timple and @JonasLos, I was able to recreate the issue locally and have a solution for you. However, this solution does not apply to all users (when used incorrectly, user may achieve adverse results), and I am hesitant to post it here on the message board. 
Would you be able to provide your email addresses for me to contact you? Or if you prefer, you can send an email to support.novatel@hexagon.com addressing Joe with a reference to this issue, and I will be able to respond promptly.

JonasLos, 2023-06-20:
my email is jonaslo96@tamu.edu

Thank you

novatel-applications-engineering, 2023-06-21:
Hi @JonasLos, I have taken the case you created. If you would like, I can delete the previous message to avoid sharing your email with the public.

novatel-applications-engineering, 2023-06-22:
For other users that also have this issue, please email support.novatel@hexagon.com with a reference to this Github issue, and we will be happy to provide the solution to you.

closed 2023-06-22 by novatel-applications-engineering