novatel/novatel_oem7_driver #78: Building `humble` branch with `ninja` fails

VRichardJP, 2023-10-25:
**Describe the bug**

I am building the driver from a ROS2 workspace with `colcon`. When I use `make` (default cmake generator), there is no problem:
```bash
sig@vr-desktop:~/autoware$ rm -rf ./build/novatel_oem7_driver; colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select novatel_oem7_driver
Starting >>> novatel_oem7_driver
[Processing: novatel_oem7_driver]
--- stderr: novatel_oem7_driver
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
Cloning into 'novatel_oem7_decoder'...
HEAD is now at d02ccc2 Updating minimum cmake version to satisfy warnings on ROS2 build farm
CMAKE Host :  Linux
CMAKE_SYSTEM_PROCESSOR:x86_64
CMAKE_LIB_SHARED:
ARCH is not Set, default to x64
ARCH is x64 & Building for 64 bit
CMAKE_C_FLAGS_DEBUG:    -g
CMAKE_CXX_FLAGS_DEBUG:  -g -std=gnu++11 -fPIC -D_LINUX_
CMAKE_C_FLAGS_RELEASE:  -s -O3 -DNDEBUG
CMAKE_CXX_FLAGS_RELEASE:-g -Wno-write-strings -Wno-format -Wno-format-security -Wno-overflow -std=gnu++11 -fPIC -D_LINUX_ -DNDEBUG -s
COMPILE_DEFINITIONS =
COMPILE_DEFINITIONS = DATADIR="/home/sig/autoware/build/novatel_oem7_driver/novatel_oem7_decoder/src/novatel_oem7_decoder/src/decoders/jsoninterface/test/resources/"
COMPILE_DEFINITIONS = DATADIR="/home/sig/autoware/build/novatel_oem7_driver/novatel_oem7_decoder/src/novatel_oem7_decoder/src/decoders/novatel/test/resources/"
COMPILE_DEFINITIONS = DATADIR="/home/sig/autoware/build/novatel_oem7_driver/novatel_oem7_decoder/src/novatel_oem7_decoder/src/hw_interface/stream_interface/test/resources/"
---
Finished <<< novatel_oem7_driver [57.3s]

Summary: 1 package finished [59.1s]
  1 package had stderr output: novatel_oem7_driver
```

But when I do the same with `ninja` generator, `ninja` complains `usr/lib/libCommon.a` and other EDIE libraries are missing:
```bash
$ rm -rf ./build/novatel_oem7_driver; colcon build --symlink-install --event-handlers=console_cohesion+ --cmake-args -GNinja -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select novatel_oem7_driver
Starting >>> novatel_oem7_driver
--- output: novatel_oem7_driver
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 1.3.5 (/opt/ros/humble/share/ament_cmake/cmake)
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found pluginlib: 5.1.0 (/opt/ros/humble/share/pluginlib/cmake)
-- Found rclcpp: 16.0.6 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.4 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "3.0.2")
-- Found FastRTPS: /opt/ros/humble/include
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found rclcpp_components: 16.0.6 (/opt/ros/humble/share/rclcpp_components/cmake)
-- Found gps_msgs: 2.0.3 (/opt/ros/humble/share/gps_msgs/cmake)
-- Found sensor_msgs: 4.2.3 (/opt/ros/humble/share/sensor_msgs/cmake)
-- Found nmea_msgs: 2.0.0 (/opt/ros/humble/share/nmea_msgs/cmake)
-- Found nav_msgs: 4.2.3 (/opt/ros/humble/share/nav_msgs/cmake)
-- Found tf2_geometry_msgs: 0.25.4 (/opt/ros/humble/share/tf2_geometry_msgs/cmake)
-- Found eigen3_cmake_module: 0.1.1 (/opt/ros/humble/share/eigen3_cmake_module/cmake)
-- Found Eigen3: TRUE (found version "3.4.0")
-- Ensuring Eigen3 include directory is part of orocos-kdl CMake target
-- Found novatel_oem7_msgs: 20.1.0 (/home/sig/autoware/install/novatel_oem7_msgs/share/novatel_oem7_msgs/cmake)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0")
-- Found GeographicLib: /usr/lib/x86_64-linux-gnu
-- System Host Proc: 'x86_64'
-- System Proc: 'x86_64'
-- VOID_P: 8
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
-- Found launch_testing_ament_cmake: 1.0.4 (/opt/ros/humble/share/launch_testing_ament_cmake/cmake)
-- Found python_cmake_module: 0.10.0 (/opt/ros/humble/share/python_cmake_module/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.6")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found suitable version "3.10.12", minimum required is "3.5")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.10
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.10.so
-- Found PythonExtra: .so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sig/autoware/build/novatel_oem7_driver
ninja: error: 'usr/lib/libCommon.a', needed by 'libnovatel_oem7_driver.so', missing and no known rule to make it
---
--- stderr: novatel_oem7_driver
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
ninja: error: 'usr/lib/libCommon.a', needed by 'libnovatel_oem7_driver.so', missing and no known rule to make it
---
Failed   <<< novatel_oem7_driver [1.90s, exited with code 1]

Summary: 0 packages finished [3.68s]
  1 package failed: novatel_oem7_driver
  1 package had stderr output: novatel_oem7_driver
```

The problem here is that `ninja` tries to resolve the dependencies early, but has no way to know the `novatel_oem7_decoder` external project will build the libraries it is looking for. This problem and a workaround are described [here](https://stackoverflow.com/questions/48142082/cmake-externalproject-add-project-not-building-before-targets-that-depend-on-it).

A simple way to fix is to list all the missing libraries with `BUILD_BYPRODUCTS`, like this:
```cmake
include(ExternalProject)
ExternalProject_Add(${OEM7_DECODER}
  GIT_REPOSITORY https://github.com/novatel/novatel_edie
  GIT_TAG origin/dev-ros_install_prefix
  UPDATE_DISCONNECTED TRUE
  PREFIX ${CMAKE_BINARY_DIR}/${OEM7_DECODER}
  CMAKE_ARGS -DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_BINARY_DIR} -DVERSION=1.0.0.2
  BUILD_BYPRODUCTS 
    ${CMAKE_BINARY_DIR}/usr/lib/libCommon.a
    ${CMAKE_BINARY_DIR}/usr/lib/libStreamInterface.a
    ${CMAKE_BINARY_DIR}/usr/lib/libNovatel.a
)
```

Then, the package compiles as expected even with `ninja` generator.

**To Reproduce**
`$ colcon build --symlink-install --event-handlers=console_cohesion+ --cmake-args -GNinja -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select novatel_oem7_driver`

**Expected behavior**
Compiles

**Screenshots**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
N/A

novatel-applications-engineering, 2024-02-29:
@VRichardJP thank you for your response. Can you upload the terminal output when you are building with ninja after the changes you suggested ?

VRichardJP, 2024-02-29:
before:
```
sig@vr-desktop:~/autoware$ rm -rf ./build/novatel_oem7_driver; colcon build --symlink-install --event-handlers=console_cohesion+ --cmake-args -GNinja -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select novatel_oem7_driver
Starting >>> novatel_oem7_driver
--- output: novatel_oem7_driver                         
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 1.3.7 (/home/sig/autoware/install/ament_cmake/share/ament_cmake/cmake)
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found pluginlib: 5.1.0 (/opt/ros/humble/share/pluginlib/cmake)
-- Found rclcpp: 16.0.8 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_cyclonedds_cpp: 1.3.4 (/opt/ros/humble/share/rmw_cyclonedds_cpp/cmake)
-- Using RMW implementation 'rmw_cyclonedds_cpp' as default
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found rclcpp_components: 16.0.8 (/opt/ros/humble/share/rclcpp_components/cmake)
-- Found gps_msgs: 2.0.3 (/opt/ros/humble/share/gps_msgs/cmake)
-- Found sensor_msgs: 4.2.3 (/opt/ros/humble/share/sensor_msgs/cmake)
-- Found nmea_msgs: 2.0.0 (/opt/ros/humble/share/nmea_msgs/cmake)
-- Found nav_msgs: 4.2.3 (/opt/ros/humble/share/nav_msgs/cmake)
-- Found tf2_geometry_msgs: 0.25.5 (/opt/ros/humble/share/tf2_geometry_msgs/cmake)
-- Found eigen3_cmake_module: 0.1.1 (/opt/ros/humble/share/eigen3_cmake_module/cmake)
-- Found Eigen3: TRUE (found version "3.4.0") 
-- Ensuring Eigen3 include directory is part of orocos-kdl CMake target
-- Found novatel_oem7_msgs: 20.1.0 (/home/sig/autoware/install/novatel_oem7_msgs/share/novatel_oem7_msgs/cmake)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0")  
-- Found GeographicLib: /usr/lib/x86_64-linux-gnu  
-- System Host Proc: 'x86_64'
-- System Proc: 'x86_64'
-- VOID_P: 8
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
-- Found launch_testing_ament_cmake: 1.0.4 (/opt/ros/humble/share/launch_testing_ament_cmake/cmake)
-- Found python_cmake_module: 0.10.0 (/opt/ros/humble/share/python_cmake_module/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.6") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found suitable version "3.10.12", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.10
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.10.so
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sig/autoware/build/novatel_oem7_driver
ninja: error: 'usr/lib/libCommon.a', needed by 'libnovatel_oem7_driver.so', missing and no known rule to make it
---
--- stderr: novatel_oem7_driver
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
ninja: error: 'usr/lib/libCommon.a', needed by 'libnovatel_oem7_driver.so', missing and no known rule to make it
---
Failed   <<< novatel_oem7_driver [4.18s, exited with code 1]

[... 437 lines cut ...]
--- stderr: novatel_oem7_driver
Linking to EDIE at: '/home/sig/autoware/build/novatel_oem7_driver'
---
Finished <<< novatel_oem7_driver [1min 2s]

Summary: 1 package finished [1min 4s]
  1 package had stderr output: novatel_oem7_driver

```

diff:
```diff
--- a/src/novatel_oem7_driver/CMakeLists.txt
+++ b/src/novatel_oem7_driver/CMakeLists.txt
@@ -49,6 +49,10 @@ ExternalProject_Add(${OEM7_DECODER}
   UPDATE_DISCONNECTED TRUE
   PREFIX ${CMAKE_BINARY_DIR}/${OEM7_DECODER}
   CMAKE_ARGS -DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_BINARY_DIR} -DVERSION=1.0.0.2
+       BUILD_BYPRODUCTS
+    ${CMAKE_BINARY_DIR}/usr/lib/libCommon.a
+    ${CMAKE_BINARY_DIR}/usr/lib/libStreamInterface.a
+    ${CMAKE_BINARY_DIR}/usr/lib/libNovatel.a
 )
 
```

novatel-applications-engineering, 2024-03-08:
@VRichardJP thank you for your response. As you can notice in your terminal output, there are git failures with the following lines: 

-- Failed to determine GIT_SHA from repository tags. Using default version "0000000000000000".
-- Failed to determine GIT_BRANCH from repository tags. Using default version "Error".
-- Failed to determine GIT_IS_DIRTY from repository tags. Using default version "false".
-- Failed to determine BUILD_TIMESTAMP from repository tags. Using default version "0000-00-00T00:00:00".

We are investigating the issue and will get back to you soon.

novatel-applications-engineering, 2024-09-27:
Hi @VRichardJP 

I am sorry for the delay in this case. Thank you for making this suggestion. We have added your proposed change to our Humble branch today.

closed 2024-10-08 by novatel-applications-engineering