novatel/novatel_oem7_driver #79: PwrPak7 + ISA100C has wrong raw IMU value

sherman-luo, 2023-11-17:
/imu/data_raw has wrong values it seems

When echoing the topic, I see

linear_acceleration ~[0, 0, 4.91], which is about half what I would expect. Is there a bug here?

sherman-luo, 2023-11-17:
I found that the ISA-100C has a rate set to 100 in oem7_supported_imus.yaml when in reality it seems to be running at 200hz. That might be the reason why?

sherman-luo, 2023-11-17:
I changed the rate to 200 and it seems to fix it. The IMU model that the driver identified was "Northrop Grumman Litef ISA-100C" at 100HZ when I think that my product is actually different at 200HZ 
https://novatel.com/products/gnss-inertial-navigation-systems/imus/isa-100c

Could it be that the model I have it not officially supported by the driver yet? It seems like the Litef is not what I have despite the similar name

novatel-applications-engineering, 2023-11-20:
Thank you for reaching out to us @sherman-luo 
We have received your request, and we will start investigating the issue.

novatel-applications-engineering, 2023-11-30:
Hi @sherman-luo, ISA100C is an IMU that outputs at 200Hz, after some investigation, this is an error during initial creation of the oem7_supported_imus.yaml file. We will work on fixing it in the source code for all branches. 

Would you mind letting us know if you install NovAtel ROS driver via the Ubuntu APT package manager? If so, what is the version of NovAtel ROS driver you use? (ROS1 or ROS2? Plus the corresponding version number)

sherman-luo, 2023-11-30:
ROS1 driver
Building from source code, I'm on master from about a week ago

novatel-applications-engineering, 2023-11-30:
Thank you for the information. I will update you on the changes.

novatel-applications-engineering, 2024-02-07:
Hi @sherman-luo, thank you for your patience. We have made changes to the IMU rate for ISA 100C. It is recommended to change the ONTIME trigger of [ CORRIMUS](https://docs.novatel.com/OEM7/Content/SPAN_Logs/CORRIMUS.htm) log to get desired IMU rate. 
Changes are committed to ROS1 branch. Commit link : [https://github.com/novatel/novatel_oem7_driver/commit/fc913e573ff607118289f552f49471470f3544f2]

closed 2024-02-07 by novatel-applications-engineering