novatel/novatel_oem7_driver #80: Ungracefull shutdown of the `readMessage` call in a Nodelet

Timple, 2023-11-23:
**Describe the bug**
Running the nodelet without gps antenna attached doesn't let the nodelet shutdown.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the `Oem7MessageNodelet`
2. launch the file
3. Stop with `ctrl+c`

**Expected behavior**
Gracefull shutdown (return code 0)

**Environment (please complete the following information):**
 - ros noetic

**Additional context**
The problem seems to be in the closed-sourced `if(decoder_->readMessage(msg))` call. Therefor i cannot be of further assistance. It blocks and doesn't stop upon ros shutdown.

novatel-applications-engineering, 2023-11-24:
Hello @Timple 
Thank you for sharing your finding.
We will investigate the issue and release a fix accordingly.

novatel-applications-engineering, 2024-03-04:
Hello @Timple, after some investigation, engineering has found that the proper fix for this is not trivial. 

Would you please clarify the following?
- How important is this to you? 
- How much is this really needed?

Timple, 2024-03-04:
We have tests in place that check for proper shutdown. So we might need to maintain an exception for this node. Running code is not hindered, so it's not the most urgent.
Does the same thing occur in ROS2?

novatel-applications-engineering, 2024-03-07:
Thank you for the update. After looking through the source code for ROS2, it likely will have the same ungraceful shutdown behavior as ROS1.

novatel-applications-engineering, 2025-02-26:
Hi @Timple thanks for your patience. We have reworked our oem7_receiver_if plugin to allow for a proper shutdown when the receiver is not initially connected or if shutdown is requested after the receiver disconnects. The code should now be non-blocking for the receiver interface. Please let me know if you run into any issues with it!

Unfortunately, due to this requiring major changes, we can only apply this change to the Jazzy branch and future releases, not any versions that have been previously released.

So you are aware, you had suggested that the issue occurred in novatel_oem7_decoder. While the code does go there, it then loops back novatel_oem7_driver for the receiver interface, so these issues were resolved through modifications to only novatel_oem7_driver.

For your general information, the decoder is also open source!
[https://github.com/novatel/novatel_edie/tree/dev-ros_install_prefix](https://github.com/novatel/novatel_edie/tree/dev-ros_install_prefix)

Timple, 2025-02-27:
I did not know the decoder was also open source. Nice.
If you think this issue is fixed, feel free to close it.

I can always re-open if I notice otherwise, but I don't have the time for testing right now.

novatel-applications-engineering, 2025-02-27:
Thanks @Timple. Yes, please let us know if you run into any of these issues again in Jazzy, but from our testing it is resolved.

closed 2025-02-27 by novatel-applications-engineering