novatel/novatel_oem7_driver #81: Unclear what frames mean in ROS messages

sherman-luo, 2023-11-29:
For the following output types and frame_ids, I am unclear what the frames represent
```
IMU:        {topic: /gps/imu,                 frame_id: imu_link}
RAWIMU:     {topic: /imu/data_raw,            frame_id: imu_link}
GPSFix:     {topic: /gps/gps,                 frame_id: gps}
NavSatFix:  {topic: /gps/fix,                 frame_id: gps} 
Odometry:   {topic: /novatel/oem7/odom,       frame_id: odom} 
```

Can we be clear what transformations each of the messages represent? 

I assume `gps` means it's the pose of the gps in world frame, in other words the transformation from the gps coordinate frame on the vehicle to the world UTM frame. 

I am most unclear about what `odom` means here. What is the `odom` frame? The pose inside `/novatel/oem7/odom` I assume represents the transformation from the `odom` frame to the world UTM frame. But what is that `odom` frame?

novatel-applications-engineering, 2023-11-30:
Hi @sherman-luo, thank you for your patience on a response.

Though BESTPOS and INSPVA/S/X come from different frames, the position solution from both NovAtel messages are in the same coordinate system. For example, Single Point position will report in WGS84 coordinate system. PPP position will report in ITRF2014 . RTK position will report in the coordinate system and datum used by the base station, and rover (unit in your vehicle) does not know this information. 

The main difference between BESTPOS and INSPVA/S/X position solution is:
1. BESTPOS position solution (best available solution, could be either GNSS only or INS solution) is output at the antenna phase center
2. INSPVA/S/X position solution (INS solution) is by default output at the IMU center of navigation (as labelled on the enclosure). INS solution output location can be translated to another location by using SETINSTRANSLATION USER command. If you give it the primary antenna lever arm (same values used in SETINSTRANSLATION ANT1 command), INS solution output location will be translated to the antenna phase center. 

UTM position solution from NovAtel receiver is transformed from BESTPOS solution (best available GNSS only or INS solution), so odometry topic solution output location is at the antenna phase center.

As described in [REP-105](https://ros.org/reps/rep-0105.html#coordinate-frames), odom frame is world fixed frame. In the NovAtel ROS driver, odometry topic position is output in UTM coordinates to be compliant with REP-105. 

Please let me know if you have any questions.

sherman-luo, 2023-11-30:
Thanks for the reply. I think all makes sense except this line

"UTM position solution from NovAtel receiver is transformed from BESTPOS solution (best available GNSS only or INS solution), so odometry topic solution output location is at the antenna phase center."

But I believe that Odometry topic position could be either from BESTPOS  **or** INSVPA. That means that the Odometry topic position could be the output from either the antenna phase center **or** the SPAN (Default IMU) frame, correct?

sherman-luo, 2023-12-06:
@novatel-applications-engineering in case you didn't see this

novatel-applications-engineering, 2023-12-07:
Hi @sherman-luo, thank you for your patience on a response.

That is correct. My apologies, I had a typo in the previous response (I have edited the previous response for future reference). Depending on the source selected by the ROS driver (by default it will toggle between BESTPOS or INSPVA), the odom position output can switch between two places, antenna phase center or IMU center of navigation. The solution to get around this issue:

1. Select preferred position source to be INS in the ROS driver, this way the position solution will always be the INS position solution output location, which by default is IMU center of navigation and it can be translated to another location.
2. (Recommended) Translate the INS position solution to the antenna phase center by adding SETINSTRANSLATION USER command with primary antenna lever arm values in the std_init_commands.yaml file. This will ensure the position solution is always output at antenna phase center regardless if the position solution toggles between BESTPOS or INSPVA. Please see below for links to references:
https://wiki.ros.org/novatel_oem7_driver/configuration
https://docs.novatel.com/OEM7/Content/SPAN_Commands/SETINSTRANSLATION.htm

I hope this helps. Please let me know if you have any questions.

closed 2023-12-07 by sherman-luo