novatel/novatel_oem7_driver #82: Requesting support for ROS2 Iron

RichardvdK, 2024-01-02:
The driver fails to install in the latest ROS2 Iron distro. Is there a possibility to get an Iron release?

Error message: `E: Unable to locate package ros-iron-novatel-novatel_oem7_msgs`

**Describe the solution you'd like:**
- A ROS2 Iron release

mbharatheesha, 2024-02-02:
I can confirm that the driver builds on `Iron`. There's just one small change needed to specify the QoS settings which would otherwise cause a deprecation warning during compilation. After making the QoS profile changes, I can confirm that the code compiles without any warnings.

Other than that change, I tested the compiled code on Iron with a receiver, using the default launch files in this repo as well as some custom launch files and custom receiver init commands I needed for our project. It all works as expected.

@novatel-applications-engineering Do I target the changes to `master` branch or `ros2-dev` branch? Or will there be a `iron` branch soon?

novatel-applications-engineering, 2024-02-05:
Hi @RichardvdK and @mbharatheesha there are no plans to release the `iron` branch as of now. @mbharatheesha please target the changes to the `humble` branch, which is our latest release. Kindly ignore the `humble-dev` branch.

closed 2025-04-09 by novatel-applications-engineering