novatel/novatel_oem7_driver #87: Recording Bag with novatel_oem7_driver Results in Inconsistent Data Logging

QamarBader, 2024-04-30:
When recording a bag using the novatel_oem7_driver, I've encountered an issue where the recorded frames exhibit intermittent gaps followed by bursts of frames, as illustrated in the screenshot attached. This erratic behavior disrupts the continuity of data and could potentially impact downstream processes relying on a consistent data stream. I tried to re-write the bag with header time but the problem persisted. 

[image]

**Steps to Reproduce**: 
`roslaunch novatel_oem7_driver oem7_net.launch`
`rosbag record -a`

**Environment**:
 - OS: Ubuntu
 - Version: 20.04.6 LTS
 - ROS DISTRO: noetic

JonasFrey96, 2024-06-28:
We are facing here the same problem. 
We performed some analysis on the frequencies of different outputs where we plot the delta t difference between to consecutive messages in a histogram in Hz. We would like to see a dirac at the given rate instead of the following:
[image]

novatel-applications-engineering, 2024-07-07:
It appears that Unix time is being used for plotting between messages, instead of the GPSTime available in the message headers. Please note that Unix time differs from GPSTime.

Plotting Unix timestamps is not ideal for confirming data inconsistency, since Unix Time reflects driver execution. Plotting GPSTime available in the Nov Header is more appropriate.

For example, if you are logging the Bestpos log at 20Hz (T=1/F), the GPSTime difference between the second and first message should be 0.05 seconds (50 milliseconds)

closed 2024-07-07 by novatel-applications-engineering

JonasFrey96, 2024-07-08:
Dear Novatel, 
Thanks a lot again for the answer. This perfectly helps us out. 
Best Jonas