novatel/novatel_oem7_driver #90: Requesting support for ROS2 Jazzy

Timple, 2024-05-21:
**Is your feature request related to a problem? Please describe.**
When we switch to ROS2 jazzy, we would like the novatel binaries to be available

**Describe the solution you'd like**
Release the `novatel_oem7_driver` to ROS2 Jazzy buildfarm

**Describe alternatives you've considered**
-

**Additional context**
Similar to: https://github.com/novatel/novatel_oem7_driver/issues/82, but since ROS2 jazzy is an [LTS](https://dlu.github.io/ros_clock/index.html) release, I assume this is targetted.

novatel-applications-engineering, 2024-09-25:
Hi Timple. We currently have a branch jazzy-dev under development for this. We are working on making some improvements to address user issues prior to releasing jazzy.

novatel-applications-engineering, 2025-02-26:
Hi @Timple thanks for your patience. Jazzy is now officially released on the jazzy branch. We are in the process of releasing it to the ROS build farm. I will update here once it can be installed from the build farm through apt.

Timple, 2025-02-27:
Good to hear!

And thank you for also releasing!

While you're on it, perhaps releasing to `rolling` would be an option. That way all future ROS distributions are immediately supported instead of 1 year after release 😉

closed 2025-02-27 by Timple

novatel-applications-engineering, 2025-02-27:
Thanks @Timple for the suggestion. We are looking at your request internally to consider a release to rolling.

novatel-applications-engineering, 2025-03-11:
Hi @Timple, our package for Jazzy is now fully released and can be installed via apt.

I have passed your feedback for a rolling release along internally. They are still looking into it.