novatel/novatel_oem7_driver #91: Sending ASCII commands to CPT7

JonasFrey96, 2024-06-28:
Hi NovAtel, 

it would be great if a service exists that allows one to directly send ASCI commands to the CPT7 via a service. 
One e.g. want to start via ROS the recording of topics 

```
"FILECONFIG OPEN"
"LOG FILE BDSEPHEMERIS",
"LOG FILE BESTGNSSPOS",
"LOG FILE BESTPOS",
"LOG FILE GALINAVRAWEPHEMERIS",
"LOG FILE INSCONFIG",
"LOG FILE INSPVAX",
"LOG FILE INSUPDATESTATUS",
"LOG FILE IONUTC",
"LOG FILE RANGECMP4",
"LOG FILE RAWEPHEM",
"LOG FILE RAWIMUSX",
"LOG FILE RXSTATUS"
```

This would be significantly easier for robotics people to just call a service and receive a response for each command one would like to send, which contains the correct ASCII command. 
Currently we are running the following python script - however this fails to decode the response and seems to be not reliable. 
```python
import socket

# Define the IP address and port
IP = "192.168.2.98"
PORT = 3001

# Define the command to be sent
command = f"SETAPPROXTIME {gps_weeks} {round(gps_seconds)}"

def send_cmd(command):
    try:
        # Create a socket object
        with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
            # Connect to the server
            s.connect((IP, PORT))
            print(f"Connected to {IP}:{PORT}")
            # Send the command
            s.sendall(command.encode('utf-8'))
            print(f"Command sent: {command}")
            # Receive response (if any)
            response = s.recv(1024)        
            # Try to decode the response as UTF-8
            print(response[:7].decode('utf-8'))
    except Exception as e:
        print(f"An error occurred: {e}")

send_cmd(command)
send_cmd("LOG TIME")
```

Best regards, Jonas

novatel-applications-engineering, 2024-07-04:
Hi @JonasFrey96 

Please note that there is an option to send Abbreviated ASCII commands directly to the receiver via ROSSERVICE. More information can be found in the _Runtime OEM7 Reconfiguration_ section here [novatel_oem7_driver/runtime_operation - ROS Wiki](https://wiki.ros.org/novatel_oem7_driver/runtime_operation#Runtime_OEM7_Reconfiguration).

closed 2024-07-05 by novatel-applications-engineering

JonasFrey96, 2024-07-05:
Amazing, thanks a lot - we were not aware of that feature. 
Best Jonas