novatel/novatel_oem7_driver #94: How to verify if the GPS signal has RTK correction applied?

GPrathap, 2025-02-20:
Is the correct approach to determining whether RTK correction was applied to the received GPS signal unclear?

GPrathap, 2025-02-20:
I guess it is related here : https://github.com/novatel/novatel_oem7_driver/issues/58

GPrathap, 2025-02-21:
Hi @novatel-applications-engineering  any update on this?

novatel-applications-engineering, 2025-02-21:
Hello @GPrathap 
Can you please clarify this question?  I read this as: what is the best way to determine whether or not RTK corrections have been applied to the GNSS solution?

If this is the question, there are a few ways you can achieve this.

- You can monitor the "diff_age" field, position solution status field, in BESTPOS ( https://docs.novatel.com/OEM7/Content/Logs/BESTPOS.htm ) or BESTGNSSPOS ( https://docs.novatel.com/OEM7/Content/SPAN_Logs/BESTGNSSPOS.htm )

- If the age is greater than 0, you are receiving corrections, and it should increase with each epoch

- The GPSFix topic ( https://docs.ros.org/en/humble/p/gps_msgs/interfaces/msg/GPSFix.html ) can be monitored as well for the GPSStatus field to show the measurement status of RTK_FIX or RTK_FLOAT

If I misunderstood the question, please let me know.

GPrathap, 2025-02-21:
Hi there, thanks for the reply, sometime due to some unknown reasons if we loose the mount point for rtk correction, we need to indicate such that we don't have rtk correction for current reading, yes we get the pose from `bestpos`

GPrathap, 2025-02-21:
Let us know you understand our problem 🙂

novatel-applications-engineering, 2025-02-24:
Hi @GPrathap ,
There are a few ways that you can monitor if the RTK corrections are being used.

- If you watch the RXSTATUS log, you can watch bit 14 and bit 15 to see if there are usable corrections https://docs.novatel.com/OEM7/Content/Logs/RXSTATUS.htm

- The diff_age field in the BESTPOS log shows you the age of the differential corrections. From this you could set a threshold in your system. Additionally, a differential age of 0 indicates that no differential correction was used.
- I had to update my original post response because my statement was incorrect, so just calling attention to that here. 
- The RTKPOS log contains the Low-Latency RTK position when valid, and an "invalid" status when a Low-Latency RTK solution could not be computed. ( https://docs.novatel.com/OEM7/Content/Logs/RTKPOS.htm )

Let me know if you have any other questions.

GPrathap, 2025-02-24:
Hi @novatel-applications-engineering, thank you very much for the detailed explanation.  

1. `/novatel_driver/rxstatus` 

              aux4_stat_bits: [0, 14, 21]
              aux4_stat_strs: [<60% of available satellites are tracked well, 'Usable RTK Corrections: < 60%', Poor
                  ALIGN COM link]

 2.  `/novatel_driver/bestpos ` 
 
              undulation: 7.5
              datum_id: 34
              lat_stdev: 0.0308949109167
              lon_stdev: 0.0408235453069
              hgt_stdev: 0.0531800501049
              stn_id: "9\0\0\0"
              diff_age: 1.14999997616
              sol_age: 0.0
              num_svs: 30
              num_sol_svs: 27
              num_sol_l1_svs: 27
              num_sol_multi_svs: 18
              reserved: 0
              ext_sol_stat: 
                status: 33
              galileo_beidou_sig_mask: 53
              gps_glonass_sig_mask: 51    

3. I could not find this `RTKPOS` is this another ros topic? can you help me with this, please?

novatel-applications-engineering, 2025-02-24:
Hi @GPrathap,
My apologies, the RTKPOS is a standard NovAtel log, but not one included in our ROS driver. 
The BESTPOS differential age and RXSTATUS monitoring will be your best options at this time.

Thanks for the understanding

closed 2025-03-11 by novatel-applications-engineering

GPrathap, 2025-03-21:
@novatel-applications-engineering  

1. We are having some problems. Is it possible to get 2d_SD through the ros driver?

[image]

2. `The diff_age field in the BESTPOS log shows you the age of the differential corrections. From this you could set a threshold in your system. Additionally, a differential age of 0 indicates that no differential correction was used.`

However, we have noticed that if `diff_age` is higher we don't have proper RTK correction. 

Please correct us?

GPrathap, 2025-03-24:
@novatel-applications-engineering  is it possible to get quick response regarding this?

novatel-applications-engineering, 2025-03-25:
Hi @GPrathap,
Apologies for the delay.  
**_Q1: Is it possible to get 2d_SD through the ros driver?_**
2D Standard Deviation can be calculated from the lat, lon standard deviation in the BESTPOS log. 
**_Q2: The diff_age field in the BESTPOS log shows you the age of the differential corrections. From this you could set a threshold in your system. Additionally, a differential age of 0 indicates that no differential correction was used.
However, we have noticed that if diff_age is higher we don't have proper RTK correction._**

Yes, both statements are true.  When the value is equal to zero, there are no differential corrections being used, when the time is growing, that is the age of the last used correction. So, as you noted, when diff_age is higher, the correction is older. 

Please let me know if this clears things up.